rust-osdev/uefi-rs #483: Add a `File::get_boxed_info_in` using `allocator_api`

benpye, 2022-08-20:
It would be nice if we had a version of `get_boxed_info` that did not rely on the global allocator for use with the experimental feature `allocator_api`. I think there is sufficient support to return a boxed value with a specific allocator, specifically [Box::from_raw_in](https://doc.rust-lang.org/std/boxed/struct.Box.html#method.from_raw_in).

phip1611, 2022-11-20:
@nicholasbishop I think we can solve this easily once #559 is merged.

phip1611, 2022-11-24:
Btw, as we do not have a internal discussion platform, I do it here: @nicholasbishop did you notice that I've pinged you on https://gitter.im/rust-osdev/Lobby? It'd be great if you can help

@nicholasbishop @GabrielMajeri
Would it make sense to have a dedicated communication channel (signal, Telegram) or are discussions on github sufficient?

GabrielMajeri, 2022-11-25:
> @nicholasbishop @GabrielMajeri Would it make sense to have a dedicated communication channel (signal, Telegram) or are discussions on github sufficient?

Sorry, I haven't been able to catch up and help with `uefi-rs` lately because of schoolwork and some personal projects. We can use other communication channels, sure, but I'd be just as unresponsive there as I'm currently on GitHub 😅  so I don't necessarily see the point.

nicholasbishop, 2022-11-25:
Thanks for the heads up, I didn't see that discussion on gitter.

Re a different communication channel, what about enabling [Github Discussions](https://docs.github.com/en/discussions) on the repo? It's kinda similar to issues, but can encourage broader discussion than issues which tend to just get used for bug reports.

I prefer that over chats like gitter/discord/zulip, especially because it's searchable (gets indexed by Google etc) and makes it easier to respond when one has time.

phip1611, 2022-11-25:
Actually, I just wasn't sure where totally unrelated things like "please look on gitter.im" and similar should be. I also assume such messages are very rare. Yeah, maybe the github discussions feature is what we need and a chat group overkill + has other disadvantages

PR #584 (2022-11-29): Added `read_entry_boxed_in` and `get_boxed_info_in` that use the `allocator_api`
This closes #483. It builds on the foundation created in #559.

It adds the methods `read_entry_boxed_in` and `get_boxed_info_in` that need the new `unstable_alloc` feature. `Dir::read_entry_boxed` and `File::get_boxed_info` will work with both, stable and nightly Rust (i.e., default allocation and allocator api).

**However, there is one problem**: We need to discuss how we integrate unstable features into cargo xtask build/test.

I tested it manually, but we need to test in CI if everything builds with `unstable [... cut]
closed 2022-11-29 by nicholasbishop