rust-osdev/uefi-rs #494: CI is broken

nicholasbishop, 2022-08-27:
CI started failing when nightly switched from 2022-08-25 (748038961) to 2022-08-26 (c07a8b4e0).

Error looks like this:
```
LLVM ERROR: dwo only supported with ELF and Wasm
error: could not compile `core`
```

See https://github.com/rust-osdev/uefi-rs/runs/8046893608?check_suite_focus=true for example.

nicholasbishop, 2022-08-27:
Bisected and made a bug: https://github.com/rust-lang/rust/issues/101071

nicholasbishop, 2022-08-27:
I've put up a PR that should fix this issue: https://github.com/rust-lang/rust/pull/101088

In the meantime, `nightly-2022-08-25` can be used locally.

nicholasbishop, 2022-09-01:
I think we're all good here now as of `nightly-2022-08-31`. Filed https://github.com/rust-osdev/uefi-rs/issues/498 to consider some sort of pinning in the future to avoid such a long CI outage.

closed 2022-09-01 by nicholasbishop

phip1611, 2022-09-01:
Awesome, thanks for taking care of this!