rust-osdev/uefi-rs #498: Consider adding some kind of (optional?) toolchain pinning to the CI

nicholasbishop, 2022-09-01:
Overall the nightly channel is remarkably stable, but every now and then there's a bug that breaks the uefi-rs CI. For example: https://github.com/rust-osdev/uefi-rs/issues/494

When such a bug occurs it can take a while for things to get back to working; we have to first notice the bug, then bisect to where it fails, then file an issue in the rust repo, maybe put up a fix or wait for someone else to find a fix, wait for it to get reviewed, wait for it to get merged, and wait for the following nightly build to be created. So even under the best circumstances it will take some days to be resolved.

To minimize disruption when a thing like this occurs, it would be nice to pin to a known-working nightly release so that we can continue to merge PRs. On the other hand, it's also nice to generally test against latest nightly so that we can quickly respond to "legitimate" breakages, like when an unstable feature that we rely on is changed.

I'm thinking maybe something along the lines of a `NIGHTLY_TOOLCHAIN=nightly` in the workflow file, and adjust all the jobs to use that variable. When a breakage occurs we can temporarily set it to e.g. `nightly-2022-01-02`, and then revert back to nightly once the bug has been fixed.

phip1611, 2022-09-01:
I'm in favor of it. How about using rust-toolchain.toml?

nicholasbishop, 2022-09-01:
Thanks, I'll take a look at that. I've never used rust-toolchain before