rust-osdev/uefi-rs #500: CI is broken: "GlobalValue with DLLImport Storage is dso_local!"

nicholasbishop, 2022-09-02:
Here we go again I guess :) Hot on the heels of https://github.com/rust-osdev/uefi-rs/issues/494, it looks like compilation is broken on latest nightly again: https://github.com/rust-osdev/uefi-rs/actions/runs/2980534507

```
...
GlobalValue with DLLImport Storage is dso_local!
ptr @_ZN3log20MAX_LOG_LEVEL_FILTER17h16709ec7a7aa195bE
LLVM ERROR: Broken module found, compilation aborted!
error: could not compile `uefi-services`
```

From a very quick search I don't see an existing issue in the rust repo for this, I'll try to bisect it later and file an issue.

asensio-project, 2022-09-02:
How can I bisect it properly? I read the quick guide but it's not working for me.

asensio-project, 2022-09-03:
I opened an [issue](https://github.com/rust-lang/rust/issues/101377) for fixing this.

nicholasbishop, 2022-09-03:
Thanks for filing that bug, confirmed that the PR you identified seems to be causing the breakage. I left a comment with details of what I ran to bisect.

asensio-project, 2022-09-03:
Thanks! Do you know a possible solution or something similar?

asensio-project, 2022-09-10:
The pr it's merged and the issue closed. I'm waiting for this nightly.

asensio-project, 2022-09-10:
It's fixed by the pr.

closed 2022-09-25 by nicholasbishop