rust-osdev/uefi-rs #501: Release uefi-macros-0.8.0, uefi-0.17.0, uefi-services-0.14.0

nicholasbishop, 2022-09-02:
It's probably about time to do another release. I'd like to give https://github.com/rust-osdev/uefi-rs/pull/478 a bit more time to bake in `main` in case we notice any issues with it, but can maybe do a release in a week or two.

nicholasbishop, 2022-09-09:
Well, it's been a week, nothing much has popped up since then, and I've done some more local testing without finding problems. I plan to do the release later today.

phip1611, 2022-09-09:
great, thanks!

phip1611, 2022-09-09:
oh @nicholasbishop , should we merge these two sections in the changelog?

The second one isn't a fix and the first one is a breaking change

[image]

nicholasbishop, 2022-09-09:
https://github.com/rust-osdev/uefi-rs/pull/512

nicholasbishop, 2022-09-09:
https://github.com/rust-osdev/uefi-rs/pull/513

closed 2022-09-09 by nicholasbishop