rust-osdev/uefi-rs #523: Unable to call exit_boot_services multiple times

ori-sky, 2022-10-03:
The current package API does not allow `SystemTable::exit_boot_services` to be called more than once. However, the UEFI specification requires applications to call this routine again if the first call returns `EFI_INVALID_PARAMETER`.

From https://uefi.org/specs/UEFI/2.10/07_Services_Boot_Services.html#efi-boot-services-exitbootservices:

> A UEFI OS loader must ensure that it has the system’s current memory map at the time it calls ExitBootServices(). This is done by passing in the current memory map’s MapKey value as returned by [EFI_BOOT_SERVICES.GetMemoryMap()](https://uefi.org/specs/UEFI/2.10/07_Services_Boot_Services.html#id15) . Care must be taken to ensure that the memory map does not change between these two calls. It is suggested that GetMemoryMap() be called immediately before calling ExitBootServices(). If MapKey value is incorrect, ExitBootServices() returns EFI_INVALID_PARAMETER and GetMemoryMap() with ExitBootServices() must be called again. Firmware implementation may choose to do a partial shutdown of the boot services during the first call to ExitBootServices(). A UEFI OS loader should not make calls to any boot service function other than Memory Allocation Services after the first call to ExitBootServices().

The uefi-rs documentation seems to suggest that the UEFI memory allocator cannot be used after the first call to the routine, but the above section goes contrary to this. However, as it stands, `exit_boot_services` consumes the `SystemTable<Boot>` regardless of success/failure.

Is there any chance the API could be updated to allow for the reuse of the system table, perhaps returning a new implementation of `SystemTable` through the error data that exposes a limited subset comprising just the memory allocation services and `exit_boot_services`?

nicholasbishop, 2022-10-03:
Our `exit_boot_services` wrapper internally handles retrying if `EFI_INVALID_PARAMETER` is returned: https://docs.rs/uefi/latest/src/uefi/table/system.rs.html#174

So you should be able to call it once and trust that it will either return success, or some unexpected error that isn't covered by the spec (so not `EFI_INVALID_PARAMETER`).

I think that fully addresses your question, but please feel free to reopen if I've missed anything :)

ori-sky, 2022-10-03:
Ahh perfect, this does address my question, thanks!

nicholasbishop, 2022-10-03:
It occurs to me that we should update the documentation on that method to make it clearer that it handles this for you; reopening this ticket as a reminder to do that.

ori-sky, 2022-10-03:
Ah my apologies, I've taken a closer look at the referenced code and I'm not certain that this fully resolves my concerns.

From what I can tell, internally `exit_boot_services` loops until it no longer receives an `EFI_INVALID_PARAMETER` result. However, this doesn't seem to allow for the opportunity to increase the allocation size if the memory map has gotten bigger since `memory_map_size` was previously called, and so if this situation does occur, it seems that the retrieval of the memory map could overflow the buffer passed to `exit_boot_services`. 

The documentation on `exit_boot_services` does make note that a larger size should be allocated than needed in order to account for this case, but this doesn't seem to be a guarantee, and from what I can tell from the UEFI spec (making reference again to the snippet in the original ticket above), it does seem that memory allocation services are still required to be available and could be used in conjunction with re-fetching the memory map size within the internal loop, in order to eliminate this possibility of an buffer overflow.

GabrielMajeri, 2022-11-02:
What would be the best solution in this case?

I see two possibilities for now:

- Stop taking a memory buffer parameter in the `exit_boot_services` function, and just allocate memory directly using the UEFI API (possibly re-allocating if the memory map gets bigger after the first exit call).
  This is completely safe, since [the spec guarantees that the memory allocation methods will still be available after the first call to `ExitBootServices`](https://uefi.org/specs/UEFI/2.10/07_Services_Boot_Services.html#efi-boot-services-exitbootservices):
  
  > A UEFI OS loader should not make calls to any boot service function other than Memory Allocation Services after the first call to ExitBootServices().

- Somehow return from the function without taking ownership of the `SystemTable`, and allow the users to handle this situation by themselves. In this case, the `exit_boot_services` call should become unsafe, since we can't prevent the user from making calls to boot services which the spec doesn't guarantee to be valid anymore.

We could also go for some sort of hybrid solution, where we provide an `exit_boot_services_unsafe` method (which does no error handling, but allows the caller to manage the memory buffer as they wish) and update the `exit_boot_services` function to allocate its own memory.

nicholasbishop, 2023-02-05:
I put up a PR to implement the first solution, having the function do its own allocation (with direct calls to allocate_pool/free_pool so it doesn't rely on having a Rust allocator available). Hopefully that should be sufficient for all use cases.

closed 2023-02-11 by nicholasbishop