rust-osdev/uefi-rs #524: OpenProtocolAttributes missing EFI_OPEN_PROTOCOL_BY_HANDLE_PROTOCOL

ori-sky, 2022-10-04:
The `OpenProtocolAttributes` enum is currently missing a representation of the `EFI_OPEN_PROTOCOL_BY_HANDLE_PROTOCOL` value defined by UEFI. Without this, there doesn't seem to be a way to directly open a protocol from a specified handle non-exclusively. Some example code:

```rust
let boot_services = system_table.boot_services();

let search = SearchType::from_proto::<GraphicsOutput>();
let handle_buffer = boot_services.locate_handle_buffer(search)?;

for handle in handle_buffer.handles() {
    let params = OpenProtocolParams {
        handle: *handle,
        agent: image,
        controller: None
    };
    let attribs = unsafe { core::mem::transmute(1) };
    let gop = unsafe {
        boot_services.open_protocol::<GraphicsOutput>(params, attribs)?
    };

    let modeinfo = gop.current_mode_info();
    let (width, height) = modeinfo.resolution();
    uefi_services::println!("Graphics mode: {}x{}", width, height);
}
```

When attempting to open the `GraphicsOutput` protocol exclusively here, an error seems to be encountered, presumably either upon opening the protocol or somewhere in `uefi_services::println!`, so non-exclusivity seems to be required here. However, the code succeeds when performing the unsafe `transmute`.

nicholasbishop, 2022-10-04:
You can use [`OpenProtocolAttributes::GetProtocol`](https://docs.rs/uefi/latest/uefi/table/boot/enum.OpenProtocolAttributes.html#variant.GetProtocol) for that. See https://github.com/rust-osdev/uefi-rs/blob/main/uefi-test-runner/src/proto/console/gop.rs#L5 for an example of this in action.

ori-sky, 2022-10-04:
It looks like `GetProtocol` will work in this circumstance so this ticket probably no longer affects me, although more generally `GET_PROTOCOL ` does have the caveat that any notification events created to be signaled upon a protocol being reinstalled will not be notified in the case of the protocol having been opened via `GET_PROTOCOL` -- only if it has been opened via `BY_HANDLE_PROTOCOL`, if I'm understanding the UEFI spec correctly.

JeffLi01, 2024-06-18:
I question whether anyone has been monitoring this issue, as the last comment was posted two years ago. Regardless, I would like to share my perspective.

> only if it has been opened via `BY_HANDLE_PROTOCOL`, if I'm understanding the UEFI spec correctly.

As per the UEFI specification, `BY_HANDLE_PROTOCOL` is intended for [use within the `HandleProtocol()` function of the EFI_BOOT_SERVICES](https://uefi.org/specs/UEFI/2.10/07_Services_Boot_Services.html#efi-boot-services-openprotocol:~:text=value%20is%20used.-,BY_HANDLE_PROTOCOL,-Used%20in%20the). Its name suggests that it is meant to be employed specifically for this purpose.

In my view, the existance of attribute `BY_HANDLE_PROTOCOL` is intended to give the new API `OpenProtocol()` a chance to simulate the behavior of the OLD API `HandleProtocol()`.

And the implemention in edk2 processes `BY_HANDLE_PROTOCOL` and `GET_PROTOCOL` the SAME way, meaning that there is no notification upon uninstallation/reinstallation of the protocol having been opened using `BY_HANDLE_PROTOCOL`.

Therefore, I concur with the [comment](https://docs.rs/uefi/latest/uefi/table/boot/enum.OpenProtocolAttributes.html) for `enum OpenProtocolAttributes`:

> // ByHandleProtocol (0x01) excluded because it is only intended to be
> // used in an implementation of `HandleProtocol`.

nicholasbishop, 2024-06-19:
> I question whether anyone has been monitoring this issue

I wasn't actively paying attention to this issue, but I do I get email notifications for all comments in the repo :)

I'll go ahead and close, it seems like `GetHandle` is sufficient.

closed 2024-06-19 by nicholasbishop