rust-osdev/uefi-rs #530: [SERIAL] Failed to read host reply, DEVICE_ERROR / TIMEOUT

zhang-ray, 2022-10-12:
Dear developers, I loop-back TX/RX of the only serial port of my computer, and try to send and receive some bytes to the only serial port. But I got `Failed to read host reply, DEVICE_ERROR`  (sometimes are `TIMEOUT`) error . 

Here's my code sample:

```
//# cargo build -Z build-std=core,alloc --target x86_64-unknown-uefi

#![no_std]
#![no_main]
#![feature(abi_efiapi)]

#[macro_use]
extern crate log;
#[macro_use]
extern crate alloc;

use alloc::string::String;
use uefi::prelude::*;
use uefi::proto::console::serial::Serial;
use uefi::table::boot::ScopedProtocol;
use uefi_services::{print, println};

use crate::alloc::boxed::Box;
use crate::alloc::vec::Vec;

#[entry]
fn efi_main(image: Handle, mut st: SystemTable<Boot>) -> Status {    
    uefi_services::init(&mut st).expect("Failed to initialize utilities");
    let mut bt = st.boot_services();
    let serial_handles = bt
        .find_handles::<Serial>()
        .expect("Failed to get serial handles");

    println!("[SERIAL] nb of serial: {}", serial_handles.len());

    let serial_handle = *serial_handles
        .get(0)
        .expect("Second serial device is missing");

    let mut serial = bt
        .open_protocol_exclusive::<Serial>(serial_handle)
        .expect("Could not open serial protocol");

    // Set a large timeout to avoid problems with Travis
    let mut io_mode = *serial.io_mode();
    println!("[SERIAL] original io_mode.receive_fifo_depth = {}", io_mode.receive_fifo_depth);
    println!("[SERIAL] original io_mode.timeout = {}", io_mode.timeout);
    println!("[SERIAL] original io_mode.data_bits = {}", io_mode.data_bits);
    println!("[SERIAL] original io_mode.stop_bits = {:?}", io_mode.stop_bits);
    /*
    let nb_second = 10;
    io_mode.timeout = nb_second * 1_000_000; // microseconds
    io_mode.baud_rate = 115_200;
    io_mode.data_bits = 8;
    io_mode.stop_bits = uefi::proto::console::serial::StopBits::One;
    serial.set_attributes(&io_mode).expect("Failed to configure serial port timeout");
     */
    let old_ctrl_bits = serial
        .get_control_bits()
        .expect("Failed to get device control bits");
    println!("[SERIAL] old_ctrl_bits={:?}", old_ctrl_bits);
    /*
    serial.set_control_bits(uefi::proto::console::serial::ControlBits::empty()).expect("Failed to set device control bits"); 
    */
    
    println!("[SERIAL] writing begin");
    match serial.write(make_pack_query_to_send().as_slice()) {
        Ok(__)=>{
            println!("[SERIAL] {:?}", __);
        },
        Err(e)=>{
            println!("[SERIAL] Failed to write");
            return Status::TIMEOUT;
        }
    }
    println!("[SERIAL] writing end");
    
    {
        // let mut buf = [0u8; 1<<10];
        let mut buf = [0u8; 1<<4];
        let mut found_head_byte = false;
        let mut big_buffer : Vec<u8> = Vec::new();
        let mut un_escaped_buffer : Vec<u8> = Vec::new();
        
        let if_no_last_then_zero = |pos_last_0xfe: Option<usize>| {
            match pos_last_0xfe {
                None => 0,
                Some(_usize) => _usize
            }
        };
        
        let mut go_on = true;
        while go_on {
            let mut nb_0xfe = 0;
            println!("[SERIAL] reading begin");
            match serial.read(&mut buf[..]) {
                Ok(_)=>{
                    println!("[SERIAL] OK OK OK OK");
                    println!("[SERIAL] buf={:?}", &buf);
                },
                Err(e)=>{
                    println!("[SERIAL] Failed to read host reply, e={:?}", e);
                    return Status::TIMEOUT;
                    // continue;
                }
            }
        }
    }
    Status::SUCCESS
}
```

PS 1:  The serial send is OK (I'd tried to the other computer via serial, the other computer can receive bytes)
PS 2:  I got `TIMEOUT` (not `DEVICE_ERROR`) message sometimes 
PS 3:  It seems like `io_mode.timeout`  not working when I try to `set_attributes`

GabrielMajeri, 2022-10-27:
Hello! I think it would be helpful to clarify if:

- You are working on bare hardware, or inside QEMU (and somehow connecting the physical serial port to the VM)?
- You have tested the serial port using some other _UEFI-based_ application (to make sure it's not a firmware problem)?

nicholasbishop, 2023-03-04:
Closing since I don't think we have enough info for this to be actionable currently. Please do feel free to comment with more info if you are still running into this problem.

closed 2023-03-06 by nicholasbishop