rust-osdev/uefi-rs #540: how to load drivers?

Tim-Paik, 2022-10-30:
I want to use a usb mouse in QEMU

My current method is to manually pause before QEMU executes Bootx64.efi, and execute `load UsbMouseDxe.efi` 

Now I want to use only uefi-rs to load the driver, is there any way?
Maybe we need to implement EFI_SHELL_PROTOCOL? Is there an easier way?

PS: UsbMouseDxe.efi is from Clover EFI bootloader

nicholasbishop, 2022-10-30:
I think the way to do this is through [`BootServices::load_image`](https://docs.rs/uefi/latest/uefi/table/boot/struct.BootServices.html#method.load_image). Whereas loading an application this way will just block the current application until the child exits, I think a driver will just block until it has finished initializing itself and then return control to the current application. I haven't actually tried this though, I'm just basing it off the following parts of the spec:
* 2.1.4 UEFI Drivers: "UEFI drivers are loaded by the Boot Manager, firmware conforming to this specification, or by other UEFI applications."
* 7.4 Image Services: "in some cases a UEFI application or UEFI driver may want to programmatically load and start another UEFI image. This can be done with the EFI_BOOT_SERVICES.LoadImage() and EFI_BOOT_SERVICES.StartImage() interfaces."
* Table 7.30: "When a UEFI boot service driver’s entry point completes with EFI_SUCCESS, the image is retained in memory."

Tim-Paik, 2022-10-30:
I tried loading the image directly and then running it, nothing seems to happen and no return value, maybe it fails silently?  I think it might be related to these codes https://docs.rs/uefi/latest/src/uefi/table/boot.rs.html#831

nicholasbishop, 2022-10-30:
The TODO there is for returning extra data, but it should already propagate errors. So I don't think it's failing silently.

I took a look at edk2, and I think I see what my previous comment is missing. Here's the code they have for loading a driver: [`LoadDriver`](https://github.com/tianocore/edk2/blob/master/ShellPkg/Library/UefiShellLevel2CommandsLib/Load.c#L71). At the end they call [`ConnectAllEfi`](https://github.com/tianocore/edk2/blob/master/ShellPkg/Library/UefiShellLevel2CommandsLib/Load.c#L26), which uses [`connect_controller`](https://docs.rs/uefi/latest/uefi/table/boot/struct.BootServices.html#method.connect_controller) to actually activate the driver.

Tim-Paik, 2022-11-05:
Sorry for the late reply, haven't been able to use the computer recently (I'm a student btw)
I uploaded this repo, you can try to run it:
https://github.com/Tim-Paik/uefi-rs-testpointer

There is no other output when running directly, no matter how the mouse is moved.
However, if you want to interrupt `startup.nsh` at this interface:
[image]

And execute the following command to manually load the driver, and then run `startup.nsh`
[image]

At this point, you can try to move your mouse and you will get the following output, indicating that the application can get the state of the mouse!
[image]

So I think `BootServices::load_image` and `BootServices::start_image` do not work.

Is there something wrong with my code?

nicholasbishop, 2022-11-05:
Thanks for the example code, that's very helpful to test against. As described in https://github.com/rust-osdev/uefi-rs/issues/540#issuecomment-1296299830, you'll need to call `connect_controller` as well. Add this after calling `start_image`:

```rust
let all_handles = bt().locate_handle_buffer(SearchType::AllHandles).unwrap();
for handle in all_handles.handles() {
    // Ignore errors; not all handles will have a new driver to
    // connect.
    let _ = bt().connect_controller(*handle, None, None, true);
}
```

Tim-Paik, 2022-11-06:
This looks absolutely perfect! Thanks for your answer!
It comes down to because I'm not familiar with edk2, maybe I need to try to learn its C code later.

closed 2022-11-06 by Tim-Paik