rust-osdev/uefi-rs #557: Release 0.18.0

phip1611, 2022-11-11:
Hey, I'm not sure about our release process. But I just noticed that we already have lots of functionality. Do we want to publish everything as `0.18.0` before the next release gets way too big (breaking changes etc)?

@nicholasbishop @GabrielMajeri

nicholasbishop, 2022-11-11:
I agree it's about time for a release, I can take care of that early next week, maybe Tuesday.

phip1611, 2022-11-12:
Very nice! I'm already having enough ideas for breaking changes for `0.19`  😈 😄

GabrielMajeri, 2022-11-12:
> I'm not sure about our release process and how we plan to release certain versions

We don't have a well-defined release schedule, we publish new versions whenever we feel like it and get around to doing it 😆 that being said, we do have a [well-documented release publishing process](https://github.com/rust-osdev/uefi-rs/blob/main/PUBLISHING.md).

@phip1611 would you like to be added as an owner on the crates.io packages as well, so that you can also publish new versions of the crates?

phip1611, 2022-11-13:
@GabrielMajeri if you don't mind, sure.

GabrielMajeri, 2022-11-13:
> @GabrielMajeri if you don't mind, sure.

Done, you should be able to see some owner invites in your inbox now.

phip1611, 2022-11-13:
Thanks! Crates.io sent me 4x the same email, tho 😂

GabrielMajeri, 2022-11-13:
> Thanks! Crates.io sent me 4x the same email, tho 😂

Weird, there should've been four different invites for the four crates we have (`uefi`, `uefi-macros`, `uefi-services` and `ucs2`). Did you manage to accept each one of them?

phip1611, 2022-11-13:
Ah, now I have all invitations. But some of those were sent multiple times.

phip1611, 2022-11-14:
From <https://github.com/rust-osdev/uefi-rs/pull/534#issuecomment-1281383839> I've found that we can also safely bump the MSRV with the next release. 🎉

phip1611, 2022-11-14:
Ah, @GabrielMajeri , I just noticed that you invited everyone individually

[image]

Perhaps, it also might be smart to invite `rustos-dev/uefi`? Or is there any downside? As done here for the multiboot2 crate:
[image]

nicholasbishop, 2022-11-15:
Release done :)

closed 2022-11-15 by nicholasbishop

phip1611, 2022-11-16:
Very nice, thank you!

GabrielMajeri, 2022-11-16:
> Perhaps, it also might be smart to invite `rustos-dev/uefi`? Or is there any downside? As done here for the multiboot2 crate:

Sure, you can go ahead and experiment. I had no idea you could invite GitHub teams to own crates 😅