rust-osdev/uefi-rs #564: Better organize Workspace

phip1611, 2022-11-12:
I want to suggest the following directory structure.

```
.github
book
BUILDING.md
...
ws/
  Cargo.toml
  uefi/
  uefi-services/
  xtask/
  ...
```

This will make the directory structure much cleaner and the main Cargo.toml file shorter (do not mix workspace definitions with regular crate attributes). I can work on it, if you give me the okay. Thoughts?

nicholasbishop, 2022-11-12:
I like the idea of moving the `uefi` package out of the root. The `uefi` package would then be more like all the other packages in the repo, and as you say it would cleanly separate out the `[workspace`] part of the root Cargo.toml from the `uefi` package part.

I think I would personally prefer not to move all the packages under a new directory though, all other things being equal I prefer less nesting, and our root isn't all that crowded. So I would propose essentially `mkdir uefi && mv src uefi/src` and update `Cargo.toml` accordingly.

Ignoring hidden files, our repo root would then look like this:
```
BUILDING.md
CHANGELOG.md
CONTRIBUTING.md
Cargo.toml
LICENSE
PUBLISHING.md
README.md
book
rust-toolchain.toml
template
uefi
uefi-macros
uefi-services
uefi-test-runner
xtask
```

phip1611, 2022-11-13:
> I think I would personally prefer not to move all the packages under a new directory though, all other things being equal I prefer less nesting, and our root isn't all that crowded.

yeah, that `ws/` approach originates from my work project. But I'm fine with less nesting.

GabrielMajeri, 2022-11-14:
I think we should first draw a conclusion on the discussion started in https://github.com/rust-osdev/uefi-rs/issues/560. If we end up deciding to break the `uefi-services` crate into `logger`, `global-allocator` etc. then we'll have a lot of many small crates in many directories, and moving the packages to a subdirectory would be much nicer.

phip1611, 2022-11-14:
I think these are two relatively independent steps, aren't they?

phip1611, 2022-11-14:
So, in a next step, we can move everything into o /ws/ or whatever. Smaller commits, more PRs, better review-ability, aye? :)

phip1611, 2022-11-14:
> Yeah, sure, it's just more work to do the move in two steps, vs. moving everything to where it needs to be in a go :)

I personally prefer variant A but I'm open for whatever the team consensus will be :)

closed 2022-11-17 by nicholasbishop