rust-osdev/uefi-rs #570: Regular Expression Protocol

d-sonuga, 2022-11-14:
I would like to add support for the regular expression protocol, but I would like to know if I should be concerned with issue #554 and wait for its resolution before continuing.

phip1611, 2022-11-14:
Additionally, just as a hint, I'd like to inform you that you might want to wait until #566 and #561 are solved/merged. Otherwise, you will run into merge conflicts.

d-sonuga, 2022-11-14:
Alright.
Thanks

phip1611, 2022-11-14:
It was not my intention that you close the issue. You can keep it open and we'd love to see a contribution soon :) @d-sonuga

nicholasbishop, 2022-11-15:
Re #554, you should feel free to go ahead and implement the protocol. I don't intend for that issue to block anyone, since we'll need to audit all the existing uses anyway.

d-sonuga, 2022-11-16:
Thanks @nicholasbishop . I'll go ahead then.

d-sonuga, 2023-04-30:
Hey, I implemented the protocol, but it seems like the protocol isn't supported at all. 

I attempted to start testing like this:

`pub fn test(bt: &BootServices) {
    info!("Running regex protocol test");

    let handle = bt.find_handles::<Regex>().expect("No Regular Expression handles");
    info!("Handles!: {:?}", handle);
}`

When I run this, it panics with a "No Regular Expression handles". Is there another way of testing this?

nicholasbishop, 2023-04-30:
Confirmed, seems like although there's some code for the regular expressions in EDK2, it's not exposed in OVMF currently.

You could check on real hardware to see if anything you have supports it, but as far as CI goes we won't be able to test it.

That said, we can still add the protocol to uefi-rs even if we don't have a way to test it automatically yet.

nicholasbishop, 2023-05-02:
I imagine most Rust regex crates are UTF-8, whereas this protocol uses `CHAR16*` strings. So using the protocol, if it were available, would potentially avoid having to convert between encodings.

phip1611, 2023-05-03:
Ah yes, right.