rust-osdev/uefi-rs #593: Discussing the Use of the Assignee Feature

phip1611, 2022-11-27:
I wonder if you'd see a benefit in using the assignee feature more on GitHub. This would prevent comments such as "Ping @nicholasbishop" or "@phip1611, this is ready for the next review round".

From experience, the "reviewer"-feature is not very useful. Hence, I'd suggest the following usage of the assignee feature:

Being assigned doesn't mean that you are the owner of an issue or a MR but that you have to do something on chat item. Example:

- I create a PR and assign @nicholasbishop (and @GabrielMajeri, once you have more time again, ok?) because I expect a review
- When there is feedback and change requests, @nicholasbishop might assign the issue back to me
- Once I'm done with the feedback and updating the MR, I assign @nicholasbishop again

I think this would make it more clear who is responsible for what instead of checking the discussion sections of the issues and PRs all the time.

What do you think? @GabrielMajeri @nicholasbishop

nicholasbishop, 2022-11-27:
I think for me this wouldn't add much value because I already get email notifications for everything because I "watch" the repository with notifications set to "All Activity". And usually when I review one PR I do a quick check through the others to see if anything else is ready for review (I usually skip over PRs that have merge conflicts though since I assume the PR author will fix those).

I do sometimes miss a comment where my feedback has been requested though, and in that please do @ me if I don't respond within 24 hours or so. Of course I might still not respond immediately then if other things have taken priority, but at least I'll have the reminder email in that case :)

phip1611, 2022-11-27:
I do not see an urgent need to change anything as - I think - our collaborative work is productive and quick. I just wanted to start the discussion as at work, we use the model described above, and it works well for us.

Furthermore, I also monitor the repository and walk through my email list, this works fine as well. So your point is perfectly valid.

It might happen from time to time that I assign you because I do it automatically without thinking much about it - so don't wonder then :)

closed 2022-12-26 by phip1611