rust-osdev/uefi-rs #594: Conditionally implement `core::error::Error` for all error types

nicholasbishop, 2022-11-27:
In the same spirit as https://github.com/rust-osdev/uefi-rs/pull/587, we should implement the `Error` trait for all our other error types (e.g. these ones: https://docs.rs/uefi/latest/uefi/data_types/index.html#enums). This would be gated behind the `unstable` feature.

phip1611, 2022-12-19:
We should be able to remove the `unstable` feature in 2 months or so. Since Rust 1.65, the trait is part of `libcore` in stable Rust.

https://github.com/rust-lang/rust/commit/bf7611d55ee6e24647aefc4d1c82b1dba0164536

phip1611, 2023-03-24:
Correction, this is in the core library but not yet stable.

- https://doc.rust-lang.org/stable/src/core/error.rs.html#32-195
- https://github.com/rust-lang/rust/issues/103765

closed 2023-08-17 by phip1611