rust-osdev/uefi-rs #612: The only way to get protocol that's available in runtime is deprecated

liferooter, 2022-12-19:
I want to use GOP protocol. So I have to write something like this:

```rust
// Get GOP protocol when boot services are still avaliable
let mut gop = system_table
        .boot_services()
        .open_protocol_exclusive::<GraphicsOutput>(handle)?;

/* some code */

// Exit boot services later
let (system_table, memory_map) = exit_boot_services(handle, system_table)?;
```

I get compile error. `system_table` is borrowed by `ScopedProtocol` structure stored in `gop` variable. So I need to drop `gop` variable to be able to move `system_table`. But `ScopedProtocol` implements `Drop`. Its `drop` function closes protocol, so I have no way to use GOP in runtime without usage of deprecated functions to get raw `UnsafeCell` without unneeded auto-closing.

nicholasbishop, 2022-12-19:
This is by design. Accessing any protocol after exiting boot services is not allowed by UEFI. In other words, this isn't just a restriction in uefi-rs, it's a restriction in the UEFI Specification itself.

The idea is that by the time you exit boot services you have transferred control to an operating system, and it's up to the operating system to provide things like graphics-device access.

liferooter, 2022-12-19:
But some protocols (like GOP) are allowed to be used in runtime

nicholasbishop, 2022-12-19:
Can you point me to where that info comes from? Are you seeing it work experimentally, or is there something in the specification that indicates this is allowed?

liferooter, 2022-12-19:
Yes, I've found it in UEFI specifications. Also it's told in your docs :)

https://docs.rs/uefi/latest/src/uefi/proto/console/gop.rs.html#4

nicholasbishop, 2022-12-19:
Thanks for that link. I think that might just be a bug in our docs, so I've reopened this. Can you point me to where in the UEFI Specification it says you can use GOP at runtime? I don't see any indication of that so far.

liferooter, 2022-12-19:
Well, I've just tried to find this place, but failed. Looks like I was confused, sorry. You can use the framebuffer given from GOP, not GOP itself.

closed 2022-12-19 by liferooter

nicholasbishop, 2022-12-19:
I've put up a PR to update our docs to not claim the protocol can be used at runtime.

nicholasbishop, 2022-12-20:
> Well, I've just tried to find this place, but failed. Looks like I was confused, sorry. You can use a framebuffer given from GOP, not GOP itself.

Just FYI, I'm not at all sure that using the gramebuffer pointer after exiting boot services is allowed either. See discussion here: https://github.com/rust-osdev/uefi-rs/pull/613#discussion_r1053420892

If you do have a citation from the spec that unambiguously makes it clear this is allowed, please do let me know so I can update the doc accordingly.