rust-osdev/uefi-rs #647: Screenshot test in the GOP protocol test suite fails on Windows

timrobertsdev, 2023-01-31:
The screenshot test is hitting the timeout and panicking on native Windows. Works fine on WSL2 still. 

To Replicate:
1. Clone the `main` branch
2. Run `cargo xtask run`

What I expected to happen: 
All tests to pass

What actually happened:
```
PS D:\src\uefi-rs> cargo xtask run
    Finished dev [unoptimized + debuginfo] target(s) in 0.10s
     Running `target\debug\xtask.exe run`
cargo build --target x86_64-unknown-uefi --package uefi-test-runner --features uefi-test-runner/ci --bins --examples
    Finished dev [unoptimized + debuginfo] target(s) in 0.05s
qemu-system-x86_64 -nodefaults -device virtio-rng-pci -machine q35 -smp 4 -m 256M -vga std -device isa-debug-exit,iobase=0xf4,iosize=0x04 -drive if=pflash,format=raw,readonly=on,file=C:\Program Files\qemu\share\edk2-x86_64-code.fd -drive if=pflash,format=raw,readonly=off,file=C:\Users\tim\AppData\Local\Temp\.tmp3av6j9\ovmf_vars -drive format=raw,file=fat:rw:target\x86_64-unknown-uefi\debug\esp -drive format=raw,file=C:\Users\tim\AppData\Local\Temp\.tmp3av6j9\test_disk.fat.img -serial stdio -qmp pipe:qemu-monitor -nic user,model=e1000,net=192.168.17.0/24,tftp=uefi-test-runner/tftp/,bootfile=fake-boot-file
BdsDxe: loading Boot0001 "UEFI QEMU HARDDISK QM00001 " from PciRoot(0x0)/Pci(0x1F,0x2)/Sata(0x0,0xFFFF,0x0)
BdsDxe: starting Boot0001 "UEFI QEMU HARDDISK QM00001 " from PciRoot(0x0)/Pci(0x1F,0x2)/Sata(0x0,0xFFFF,0x0)
[ INFO]: uefi-test-runner\src\main.rs@029: Firmware Vendor: EDK II
Testing print! macro with formatting: 0b10011011 Testing println! macro with formatting: 0b10011011
[ INFO]: uefi-test-runner\src\main.rs@070: UEFI 2.7
[ INFO]: uefi-test-runner\src\boot\mod.rs@006: Testing boot services
[ INFO]: uefi-test-runner\src\boot\memory.rs@006: Testing memory functions
[ INFO]: uefi-test-runner\src\boot\memory.rs@017: Allocating some pages of memory
[ INFO]: uefi-test-runner\src\boot\memory.rs@040: Allocating a vector through the `alloc` crate
[ INFO]: uefi-test-runner\src\boot\memory.rs@051: Allocating a structure with alignment to 0x100
[ INFO]: uefi-test-runner\src\boot\memory.rs@062: Testing the `memmove` / `set_mem` functions
[ INFO]: uefi-test-runner\src\boot\memory.rs@083: Testing memory map functions
[ INFO]: uefi-test-runner\src\boot\misc.rs@009: Testing timer...
[ INFO]: uefi-test-runner\src\boot\misc.rs@011: Testing events...
[ INFO]: uefi-test-runner\src\boot\misc.rs@035: Inside the event callback
[ INFO]: uefi-test-runner\src\boot\misc.rs@048: Inside the event callback with context
[ INFO]: uefi-test-runner\src\boot\misc.rs@014: Testing watchdog...
[ INFO]: uefi-test-runner\src\boot\misc.rs@016: Testing protocol handler services...
[ INFO]: uefi-test-runner\src\boot\misc.rs@107: Installing TestProtocol
[ INFO]: uefi-test-runner\src\boot\misc.rs@087: Protocol was (re)installed and this function notified.
[ INFO]: uefi-test-runner\src\boot\misc.rs@120: Reinstalling TestProtocol
[ INFO]: uefi-test-runner\src\boot\misc.rs@087: Protocol was (re)installed and this function notified.
[ INFO]: uefi-test-runner\src\boot\misc.rs@137: Uninstalling TestProtocol
[ INFO]: uefi-test-runner\src\boot\mod.rs@016: Testing the `locate_handle_buffer` function
[ INFO]: uefi-test-runner\src\proto\mod.rs@007: Testing various protocols
[ INFO]: uefi-test-runner\src\proto\console\mod.rs@004: Testing console protocols
[ INFO]: uefi-test-runner\src\proto\console\stdout.rs@005: Running text output protocol test
[ INFO]: uefi-test-runner\src\proto\console\stdout.rs@029: UEFI standard output current mode: Some(OutputMode { index: 0, dims: (80, 25) })
[ INFO]: uefi-test-runner\src\proto\console\stdout.rs@061: # uefi-rs test runner
[ INFO]: uefi-test-runner\src\proto\console\stdout.rs@065: Cursor visibility control unavailable
[ INFO]: uefi-test-runner\src\proto\console\stdout.rs@014: - Text mode #0: 25 rows by 80 columns
[ INFO]: uefi-test-runner\src\proto\console\stdout.rs@014: - Text mode #1: 31 rows by 100 columns
[ INFO]: uefi-test-runner\src\proto\console\serial.rs@050: Running serial protocol test
[ INFO]: uefi-test-runner\src\proto\console\gop.rs@007: Running graphics output protocol test
[PANIC]: panicked at 'request failed: "Screenshot("gop_test")": TIMEOUT', uefi-test-runner\src\main.rs:151:9
Error: tests did not complete successfully
error: process didn't exit successfully: `target\debug\xtask.exe run` (exit code: 1)
```

Rust version: rustc `1.69.0-nightly (001a77fac 2023-01-30)`
QEMU Version: `QEMU emulator version 7.2.0 (v7.2.0-11948-ge6523b71fc-dirty)`

nicholasbishop, 2023-02-04:
Taking a quick look now. Looks like https://github.com/rust-osdev/uefi-rs/commit/12a7b449af637745789be711a48b2833b94fdb3e is what broke it, not sure yet why though.

nicholasbishop, 2023-02-04:
Switching back the serial IO back to a pipe seems to fix that problem:

```diff
diff --git a/xtask/src/qemu.rs b/xtask/src/qemu.rs
index 80520c8..d1f1fa0 100644
--- a/xtask/src/qemu.rs
+++ b/xtask/src/qemu.rs
@@ -520,10 +520,11 @@ pub fn run_qemu(arch: UefiArch, opt: &QemuOpt) -> Result<()> {
     cmd.arg(drive_arg);
 
     let qemu_monitor_pipe = Pipe::new(tmp_dir, "qemu-monitor")?;
+    let serial_pipe = Pipe::new(tmp_dir, "serial")?;
 
     // Open a serial device connected to stdio. This is used for
     // printing logs and to receive and reply to commands.
-    cmd.args(["-serial", "stdio"]);
+    cmd.args(["-serial", serial_pipe.qemu_arg()]);
 
     // Map the QEMU monitor to a pair of named pipes
     cmd.args(["-qmp", qemu_monitor_pipe.qemu_arg()]);
@@ -556,14 +557,11 @@ pub fn run_qemu(arch: UefiArch, opt: &QemuOpt) -> Result<()> {
     let mut child = ChildWrapper(cmd.spawn().context("failed to launch qemu")?);
 
     let monitor_io = qemu_monitor_pipe.open_io()?;
-    let child_io = Io::new(
-        child.0.stdout.take().unwrap(),
-        child.0.stdin.take().unwrap(),
-    );
+    let serial_io = serial_pipe.open_io()?;
 
     // Capture the result to check it, but first wait for the child to
     // exit.
-    let res = process_qemu_io(monitor_io, child_io, tmp_dir);
+    let res = process_qemu_io(monitor_io, serial_io, tmp_dir);
     let status = child.0.wait()?;
 
     if let Some(echo_service) = echo_service {
```

Still not sure why, but that's a simple enough fix I might just do that rather than digging in deeper.

nicholasbishop, 2023-02-04:
Should be fixed by this PR: https://github.com/rust-osdev/uefi-rs/pull/651

The PR also enables running of the qemu tests on Windows in CI, so this shouldn't break again.

closed 2023-02-06 by phip1611