rust-osdev/uefi-rs #660: Capture another Key from outside the `ScanCodes` structure

JakubGawron1, 2023-02-14:
How to check if a button outside `uefi::proto::console::input::ScanCode` has been clicked.
Suppose I have this code:
```rust
let handle_input = st
        .boot_services()
        .get_handle_for_protocol::<Input>()
        .expect("Failed to handle protocol");
let mut open_input = st
        .boot_services()
        .open_protocol_exclusive::<Input>(handle_input)
        .expect("Failed to open protocol");

let key  = open_input.read_key().expect("Failed to read key");

if let Some(scancode) = key{
/*  I can then use something like this to check if ESCAPE was clicked or if a Printable key 
      if let Key::Special(ScanCode::ESCAPE) = scancode{
        println!("ESCAPE was clicked");
        break;
}
     ...

*/

}
```
Now suppose I want to intercept key 1, which has code 0x1E according to `uefi` specification.
How to capture such a key without adding it to `ScanCodes`
Thank you in advance for your help

nicholasbishop, 2023-02-18:
The 1 key is a printable key, so you can match it like this:
```rust
if let Some(Key::Printable(key)) = key {
    if key == Char16::try_from('1').unwrap() {
        info!("the '1' key was pressed");
    }
}
```

JakubGawron1, 2023-02-19:
I hadn't thought of that before, thanks a lot

closed 2023-02-19 by JakubGawron1

nicholasbishop, 2023-02-19:
Just merged a PR to add an example there so that it's clearer for people in the future. Thanks for filling this issue 🙂