rust-osdev/uefi-rs #661: Implement sorting of memory map

JarlEvanson, 2023-02-17:
To facilitate construction of memory managers, would it be possible to implement a method to obtain a sorted memory map upon a call to 'exit_boot_services'? Since iterators can't be sorted, one first has to allocate memory to event begin to be able to sort the map.

phip1611, 2023-02-17:
Hi. Generally speaking, we are open for high-level interfaces and API simplifications. Do you have a specific implementation in mind? If so, feel free to submit a PR so we can discuss further.

JarlEvanson, 2023-02-17:
Should we keep a way to obtain an unsorted memory map, or should the memory map just always be sorted for convenience? I don't see any uses that could be better done with an unsorted memory map, so the only cost would be the time needed to sort.

phip1611, 2023-02-18:
I think, instead of a new heap-structure, which will be tricky to implement, I'd prefer a stateful-iterator that always searches for the next bigger element. Given the fact that the map will not be iterated that often and that hundreds or thousands of entries are unlikely, this should be feasible.

closed 2023-03-19 by phip1611