rust-osdev/uefi-rs #697: ci: discuss approach for less duplication

phip1611, 2023-03-17:
In the multiboot2-crate, [I've successfully experimented](https://github.com/rust-osdev/multiboot2/pull/125) with configurable reusable workflows which simplified some things for me. We also do have duplications in multiple CI jobs.

Do you see a benefit in enabling something similar in the uefi-rs crate? For example, in <https://github.com/rust-osdev/uefi-rs/blob/main/.github/workflows/rust.yml> we do have 6 times a `sudo apt install qemu-*` step.

nicholasbishop, 2023-03-19:
I'm down to reduce duplication, but I'm not sure how much code duplication there is in `rust.yml` as opposed to semantic duplication. Like it's true that many jobs install qemu and ovmf, but the details are different depending on whether the target is x86_64, ia32, aarch64, and on whether the host is Windows or Linux.

phip1611, 2023-03-24:
Mostly obsolete since https://github.com/rust-osdev/uefi-rs/pull/702

I thought, the CI caching needs more boilerplate as in here: https://github.com/rust-osdev/multiboot2/blob/main/.github/workflows/_build-rust.yml#L52

closed 2023-03-24 by phip1611