rust-osdev/uefi-rs #754: What is the bare minimum necessary to get a handle to a protocol?

felipebalbi, 2023-04-04:
Hi,

I'm attempting to add support to the DebugPortProtocol in `uefi-rs`, but whenever I try to just get a handle to the protocol, it always gives me `NOT_FOUND`. The GUID I copied directly from edk2, so it should be correct. How can I make sure that the protocol, indeed, exists in the current build (I'm using Qemu Sbsa platform) of edk2?

phip1611, 2023-04-04:
Protocols are (mostly, if not all, I think) optional functionality of the UEFI firmware that may exist during runtime. So, if your code is correct, but you don't get a handle, then the UEFI implementation doesn't provide that protocol. 

Can you tell us more about your testing setup? Which firmware do you us? OVMF inside QEMU?

felipebalbi, 2023-04-04:
Hi,

as mentioned, it runs Qemu/Sbsa platform. As for firmware, I'm running arm-trusted-firmware as described by the Sbsa readme. I have added DebugPortDxe.inf to the relevant SbsaQemu.dsc file, but still nothing. Do I need to also modify the fdf file?

closed 2023-04-04 by felipebalbi