rust-osdev/uefi-rs #759: How to implement the service binding protocols

MiSimon, 2023-04-11:
I would like to implement some of the missing network protocols.
For each of them a service binding protocol with a different guid exist, what would be the correct way of implementing the service binding protocols in a way they work with open_protocol?

The obvious way would be to create a struct for every guid but that would cause a lot of duplicates.

nicholasbishop, 2023-04-12:
Maybe we could have a macro for creating these structs, like `service_binding_protocol!(name-of-struct, guid)`

MiSimon, 2023-04-13:
This will just hide the duplicates but would be easy to use and requires no further changes, i will try and implement it that way.

phip1611, 2025-04-09:
Closing due to little value-add of keeping it open. Feel free to reopen if necessary or create a new one

closed 2025-04-09 by phip1611