rust-osdev/uefi-rs #770: Production-Ready Template Project

phip1611, 2023-04-24:
Compiling a minimal hello-word UEFI application is easy. Setting up an environment with a "run in QEMU" command and unit-tests is hard, however. Especially the last part is very challenging, as one has to mix the UEFI target with a standard target for test execution. I tried working on something that supports the simultaneous building and unit testing inside the same Cargo project.

(Either I'm totally dump or this is indeed very hard).

Can you please tell me what you think? And if yes, could we integrate it into the rust-osdev namespace?

https://github.com/phip1611/rust-uefi-template

phip1611, 2023-04-24:
For example, the bootloader crate also refrains from unit tests. Probably for the reason described above. https://github.com/rust-osdev/bootloader/blob/main/uefi/src/main.rs

blitz, 2023-04-24:
We can chat about it this week in person. :)

phil-opp, 2023-04-29:
My opinion is that unit tests should be self-contained and independent of the target system. So for target-specific crates such as UEFI applications, I would move all the platform-independent stuff to a separate crate, which can then be tested as usual without requiring any conditional compilation.

For target-specific integration tests, you typically want to test them on an (emulated) target system directly. Rust doesn't really provide much support for this yet, unfortunately. I think the best solution is to build a custom test framework, e.g. using a crate such as `inventory` to collect a set of test functions.

I'm not a fan of using cfg-gates for testing since this makes it less obvious what you're really testing. For example, I think your example test at https://github.com/phip1611/rust-uefi-template/blob/2fde9ebfd9f1d43c5a90f7d0054efa050d1d55a8/src/lib.rs#L26-L29 effectively only verifies that there is a global allocator _in Rust's standard library_.

nicholasbishop, 2023-04-29:
> (Either I'm totally dumb or this is indeed very hard).

I think you are correct that it's hard. I think there are two directions to attack the problem from:
1. Improve the template project and documentation to help guide the user through the complexity
2. Improve the ecosystem to make things easier

And these are not mutually exclusive of course.

I think the idea of a standalone template UEFI app repo is good, and better than our current [template](https://github.com/rust-osdev/uefi-rs/tree/main/template) subdirectory in uefi-rs. Keeping it in a separate repo makes it easier to include things like basic github CI and potentially a simple workspace. I agree with phil-opp that using separate packages is good for testing; keep the testable-on-the-host stuff in one package and the depends-on-uefi stuff in another package. And I'm a proponent of the [xtask pattern](https://github.com/matklad/cargo-xtask) rather than a Makefile; Makefiles and shell scripts get hard to read very quickly IMO, and easy to make non-obvious mistakes when writing too.

From the direction of improving the ecosystem, perhaps we should look at librarification of some of the code in uefi-rs used to run QEMU. Pretty much every UEFI project has some local code for finding OVMF (often just a hardcoded path, but we can do better), figuring out the args for QEMU, etc. [uefi-run](https://crates.io/crates/uefi-run) exists, but it's fairly restrictive even in the library. A new library with flexible code for launching QEMU could be very helpful for getting a new project up and running.

crawfxrd, 2025-01-04:
An `xtask` pattern is likely inevitable for a pure Rust environment, as you will need/want to compile host tools/libraries for anything more than a single application/driver. `.cargo/config.toml` is generally not usable.

It is possible to separate logic that does not depend on UEFI interfaces to a library in order to test on the host, but I am unsure how useful that will actually be; Most of my logic is implemented in the driver/app _because_ it depends on such interfaces.

The ability to easily do _any kind_ of automated testing is desirably, even if it's not unit-testing on the host. I currently have to make up manual tests that generally cannot directly test functionality because I implement functionality in drivers that has no user interaction/visibility.

phip1611, 2025-10-19:
I started this initiative but I want to close it for the following reasons:

- Since I started with this, I learned much new things and would set up projects now differently
- Since @nicholasbishop  and my time on this project are limited, we don't have the capacity to maintain that

Given that we have quite some examples, ChatGPT can help people to set up working project setup, and low-level projects in its nature are often tailored to the specific use-case, I think we should just close this.