rust-osdev/uefi-rs #774: VariableAttributes should implement Default?

RaitoBezarius, 2023-04-29:
Hi there, thank you for the awesome crate.

I am doing some variables stuff and I want to use "default" variable attributes, e.g. 0x0 I believe.

Would that be possible to add the proper derive instances for VariableAttributes?

nicholasbishop, 2023-04-29:
I'm not necessarily opposed to adding `Default`, but in practice I'm not sure how useful it would be.

For example, when setting a variable you have to decide if it will only be accessible when boot services are active, in which case it will have `BOOTSERVICE_ACCESS` in attributes, or if it will be accessible during runtime services too, in which case it will have `BOOTSERVICE_ACCESS | RUNTIME_ACCESS`. Neither one is really more "default" than the other, and I think it would be an error to pass in neither, so if the default value was 0x0 I don't think it would work with `set_variable`.

It might still be worth adding a `Default` with 0x0 bits just so that the type works with generics that have a `Default` bound. Do you have a particular example in mind where `Default` would be helpful?

RaitoBezarius, 2023-04-29:
Oh apologies, I believed that 0x0 was a valid default.
Accessible by default in both phases seems a sane default IMHO when you want to be available for pre and post exit boot services but I do get your point.

RaitoBezarius, 2023-04-29:
After experimenting with this after a while, I agree with your original statement.

I guess that `BOOTSERVICE_ACCESS | RUNTIME_ACCESS` **could** be a default, but I think we should let consumers specify it. I will close this as not planned. :)

closed 2023-04-29 by RaitoBezarius