rust-osdev/uefi-rs #775: Variable setting helpers

RaitoBezarius, 2023-04-29:
Currently, `set_variable` accepts a buffer.

It would be interesting to have the classical alternatives:

- set u64
- set uint
- set string

etc.

This would make it far easier to set variables.
Potentially, the solution should be to have proper instances to transform the stuff?

Some example of the code I wrote regarding this: https://github.com/nix-community/lanzaboote/pull/166/files#diff-f2cc7d56b3462acd7190e54877874fd277591b33b4f261e03ef17ac242d033e5R47-R64

nicholasbishop, 2023-04-29:
I think serializing the data should remain up to the application. It's not as simple as it sounds to "set u64" -- big endian or little endian? Strings are even more complicated of course -- what encoding? UCS-2 is common for UEFI APIs, but there's no requirement that it be used for string data in variables. For example, you could have utf-8 encoded JSON.

RaitoBezarius, 2023-04-29:
> I think serializing the data should remain up to the application. It's not as simple as it sounds to "set u64" -- big endian or little endian? Strings are even more complicated of course -- what [...]

So Rust provides `from_le_bytes`, `from_be_bytes`, `from_ne_bytes`, etc. I wonder if it's scope creep to have similar APIs for that? Only for common usecases, I agree with you you could have anything and the sky is the limit.

nicholasbishop, 2023-04-29:
We could, but I don't think it's worth it. For example, your current code is:
```rust
let _ = runtime_services.set_variable(
    cstr16!("StubFeatures"),
    &SD_LOADER,
    VariableAttributes::from_bits_truncate(0x0),
    &stub_features.bits().to_le_bytes()
);
```

With the additional API, it would become:
```rust
let _ = runtime_services.set_variable_u64_le(
    cstr16!("StubFeatures"),
    &SD_LOADER,
    VariableAttributes::from_bits_truncate(0x0),
    stub_features.bits()
);
```

That doesn't seem any clearer to me, so I don't think it justifies adding `set_variable_*` variations.

RaitoBezarius, 2023-04-29:
Agreed, let's close it this then, thank you for the discussion!

closed 2023-04-29 by RaitoBezarius