rust-osdev/uefi-rs #780: PcrEventInputs alloc-only method?

RaitoBezarius, 2023-04-29:
Hi there (again me), I am doing some TPM2 stuff with the library.

I see that `PcrEventInputs::new_in_buffer` is available, but requires the user to perform the allocation, it is a bit suboptimal because the required size requires using computations of the sizes (basically copying the first line in the impl).

This could be awesome with a `PcrEventInputs::new` which does the allocation for us.

RaitoBezarius, 2023-04-30:
Furthermore, I realize that `EventHeader` is not public, thus, I don't see how to get the memory size using `core::mem::size_of` unfortunately.

RaitoBezarius, 2023-04-30:
(usecase: https://github.com/nix-community/lanzaboote/pull/167/files#diff-28330b5941f5596b9c3ee9e6e0eec67e914fb049a91af5a59c6ba13aa8241e09R50-R81)

phip1611, 2023-05-03:
Thanks for the report, good catch. The allocation should happen automatically, indeed.

nicholasbishop, 2023-05-17:
> Furthermore, I realize that `EventHeader` is not public, thus, I don't see how to get the memory size using `core::mem::size_of` unfortunately.

I think the simplest thing to do here is just allocate a reasonable static size. It's OK for the buffer to be bigger than necessary, it just can't be smaller. For example: https://chromium.googlesource.com/chromiumos/platform/crdyboot/+/refs/heads/main/libcrdy/src/tpm.rs#132

That said, adding a new method to create a boxed allocation would also be good, I'll take a look at doing that.

closed 2024-07-16 by phip1611