rust-osdev/uefi-rs #800: The `start_image` function keeps on failing

memN0ps, 2023-05-12:
Hi,

I've been trying to load the Windows EFI Boot Manager using the following code but `image_start()` keeps returning `INVALID_PARAMETER` even though `image_load` did not fail. Do you have any idea why this could be happening?

```rust
/// Load the Windows EFI Boot Manager (\EFI\Microsoft\Boot\bootmgfw.efi) and return a handle to it
pub fn load_windows_boot_manager(boot_services: &BootServices) -> uefi::Result<Handle> {
    // Open a protocol interface for the currently-executing image in exclusive mode with LoadedImage protocol
    let loaded_image = boot_services.open_protocol_exclusive::<LoadedImage>(boot_services.image_handle())?;

    // Open a protocol interface for the storage device in exclusive mode with SimpleFileSystem protocol to access FAT-12/16/32 file system
    let mut file_system = boot_services.open_protocol_exclusive::<SimpleFileSystem>(loaded_image.device())?;

    // Open the root directory on a volume.
    let mut root = file_system.open_volume()?;

    // Try to open a file relative to this file.
    let mut bootmgfw_file = root.open(cstr16!(r"\EFI\Microsoft\Boot\bootmgfw.efi"), FileMode::Read, FileAttribute::READ_ONLY)?.into_regular_file().unwrap();

    // Create a buffer to store file information
    let mut file_information_buffer = [0; 128];

    // Queries some information about a file
    let bootmgfw_info = bootmgfw_file.get_info::<FileInfo>(&mut file_information_buffer).unwrap();

    // File size (number of bytes stored in the file)
    let bootmgfw_size = bootmgfw_info.file_size() as usize;
    
    // Allocates from a memory pool. The pointer will be 8-byte aligned
    let memory_pool = boot_services.allocate_pool(MemoryType::LOADER_DATA, bootmgfw_size)?;
    
    // Read the empty memory pool and form a slice with a size of bootmgfw.efi
    let bootmgfw_data = unsafe { core::slice::from_raw_parts_mut(memory_pool, bootmgfw_size) };

    // Read bootmgfw.efi and populate the memory pool
    let _bytes_read = bootmgfw_file.read(bootmgfw_data).expect("Failed to read bootmgfw.efi into memory pool");
   
    log::info!("[*] Loading image...");

    // Load an EFI image into memory and return a Handle to the image.
    let bootmgfw_handle = boot_services.load_image(
        boot_services.image_handle(),
        LoadImageSource::FromBuffer {
            file_path: None,
            buffer: bootmgfw_data,
        },
    )?;

    log::info!("[+] Image Loaded Successfully!");

    // Frees memory allocated from a pool.
    boot_services.free_pool(memory_pool)?;
    log::info!("[+] Starting image...");

    //let new_loaded_image_handle = boot_services.open_protocol_exclusive::<LoadedImage>(boot_services.image_handle())?.device();
    //boot_services.start_image(new_loaded_image_handle).expect("[-] Failed to start image");

    boot_services.start_image(bootmgfw_handle).expect("[-] Failed to start image");

    return Ok(bootmgfw_handle);
}
```

phip1611, 2023-05-12:
Hi! 

Could you, just to make sure that there is no problem with any file-related operations, use the new `uefi::fs::FileSystem` abstraction? It is only on the `main`branch right now. Even though your file-related code looks good on a first glance, it is at least worth a check.

memN0ps, 2023-05-12:
Hi,

Thanks for getting back to me. I just tried this, if that is what you meant and the same error (note `boot_services.start_image` is in the same function for convenience):

```rust
/// Load the Windows EFI Boot Manager (\EFI\Microsoft\Boot\bootmgfw.efi) and return a handle to it
pub fn load_windows_boot_manager(boot_services: &BootServices) -> uefi::Result<Handle> {
    let mut file_system = boot_services.get_image_file_system(boot_services.image_handle())?;

    // Open the root directory on a volume.
    let mut root = file_system.open_volume()?;

    // Try to open a file relative to this file.
    let mut bootmgfw_file = root.open(cstr16!(r"\EFI\Microsoft\Boot\bootmgfw.efi"), FileMode::Read, FileAttribute::READ_ONLY)?.into_regular_file().unwrap();

    // Create a buffer to store file information
    let mut file_information_buffer = [0; 128];

    // Queries some information about a file
    let bootmgfw_info = bootmgfw_file.get_info::<FileInfo>(&mut file_information_buffer).unwrap();

    // File size (number of bytes stored in the file)
    let bootmgfw_size = bootmgfw_info.file_size() as usize;
    
    // Allocates from a memory pool. The pointer will be 8-byte aligned
    let memory_pool = boot_services.allocate_pool(MemoryType::LOADER_DATA, bootmgfw_size)?;
    
    // Read the empty memory pool and form a slice with a size of bootmgfw.efi
    let bootmgfw_data = unsafe { core::slice::from_raw_parts_mut(memory_pool, bootmgfw_size) };

    // Read bootmgfw.efi and populate the memory pool
    let _bytes_read = bootmgfw_file.read(bootmgfw_data).expect("Failed to read bootmgfw.efi into memory pool");
   
    log::info!("[*] Loading image...");

    // Load an EFI image into memory and return a Handle to the image.
    let bootmgfw_handle = boot_services.load_image(
        boot_services.image_handle(),
        LoadImageSource::FromBuffer {
            file_path: None,
            buffer: bootmgfw_data,
        },
    )?;
    log::info!("[+] Image Loaded Successfully!");

    // Frees memory allocated from a pool.
    //boot_services.free_pool(memory_pool)?;
    log::info!("[+] Starting image...");

    //let new_loaded_image_handle = boot_services.open_protocol_exclusive::<LoadedImage>(boot_services.image_handle())?.device();
    //boot_services.start_image(new_loaded_image_handle).expect("[-] Failed to start image");

    boot_services.start_image(bootmgfw_handle).expect("[-] Failed to start image");

    return Ok(bootmgfw_handle);
}
```

phip1611, 2023-05-12:
I was talking about sth like that:

```rust
let mut fs = bt.get_image_file_system(image)
    .expect("Failed to retrieve boot file system");

let image_data = fs.read(cstr16!(r"\EFI\Microsoft\Boot\bootmgfw.efi")).unwrap();
// Load an EFI image into memory and return a Handle to the image.
let bootmgfw_handle = boot_services.load_image(
    boot_services.image_handle(),
    LoadImageSource::FromBuffer {
        file_path: None,
        buffer: image_data,
    },
)?;
}
```

phip1611, 2023-05-12:
>  but start_image() keeps returning INVALID_PARAMETER

hm. The spec tells 

> EFI_INVALID_PARAMETER | ImageHandle is either an invalid image handle or the image has already been initialized with StartImage

memN0ps, 2023-05-12:
Hi, thanks a lot. I'm so glad I got the help to make this work at 10.30 PM Friday night. It's possible that either the image was already loaded or the image was corrupt, which was fixed after the reboot. It's also possible my original code needed to be corrected. Thank you :)

phip1611, 2023-05-12:
Alright, I'm closing the issue. Feel free to re-open if the problem occurs again.

nicholasbishop, 2023-05-12:
Start_image can also return a status value from the executable itself. So it's possible you are successfully running the image, but it detects something wrong in the env and exits with the INVALID_PARAM error. Unfortunately it's hard to tell just from the status code whether that is the case or not.

nicholasbishop, 2023-05-12:
Here's an example of setting up a device file path and loading an image that way:
https://github.com/rust-osdev/uefi-rs/pull/793/files#diff-999f2ef6b67a300d614ceb6ce1c1e7f5c4d7e7ca7e6d2545570fafe499d190fdR92

I'm not sure if that will help here, but worth a try.

medhefgo, 2023-05-12:
From what I can tell `bootmgfw.efi` expects to be called with a valid device path (presumably to find the auxiliary files it needs to boot). If it's `NULL` or the device path is something it cannot use it will return `EFI_INVALID_PARAMETER`.

memN0ps, 2023-05-13:
Hi all,

Thank you for responding so fast. Sorry for creating an issue, but a sanity check is always helpful. As I originally predicted, the problem was the file path (`DevicePath`). I assume passing the `DevicePath` as a parameter to `LoadImageSource::FromBuffer` in my original code should work too. Running the following code boots Windows now, which is what I wanted:

```Rust
/// Load the Windows EFI Boot Manager (\EFI\Microsoft\Boot\bootmgfw.efi) and return a handle to it
pub fn load_windows_boot_manager(boot_services: &BootServices) -> uefi::Result<Handle> {

    let loaded_image_device_path = boot_services.open_protocol_exclusive::<LoadedImageDevicePath>(boot_services.image_handle())?;

    let mut storage = Vec::new();
    let mut builder = DevicePathBuilder::with_vec(&mut storage);

    for node in loaded_image_device_path.node_iter() {
        if node.full_type() == (DeviceType::MEDIA, DeviceSubType::MEDIA_FILE_PATH) {
            break;
        }

        builder = builder.push(&node).unwrap();
    }

    builder = builder
    .push(&build::media::FilePath {
        path_name: cstr16!(r"EFI\Microsoft\Boot\bootmgfw.efi"),
    })
    .unwrap();
    
    let new_image_path = builder.finalize().unwrap();

    let new_image = boot_services.load_image(
        boot_services.image_handle(),
        LoadImageSource::FromFilePath {
            file_path: new_image_path,
            from_boot_manager: false,
        },
    )?;

    return Ok(new_image);
}
```

Somewhere in `main`:

```rust
let bootmgfw_handle = boot::utils::load_windows_boot_manager(boot_services).expect("Failed to load image");
boot_services.start_image(bootmgfw_handle).expect("[-] Failed to start image");
```

Is there a Discord server for help or communication? I've mostly been reading the docs and the https://rust-osdev.github.io/uefi-rs/HEAD/ as I'm new to UEFI development.

Credits/References:

* https://github.com/rust-osdev/uefi-rs/pull/793/files#diff-999f2ef6b67a300d614ceb6ce1c1e7f5c4d7e7ca7e6d2545570fafe499d190fdR92

nicholasbishop, 2023-05-13:
> Is there a Discord server for help or communication?

There's https://gitter.im/rust-osdev/Lobby, but it's not necessarily a faster way to get assistance (personally I often forget to check it). That said, it's perfectly fine to use the uefi-rs issue tracker as you have done here to ask for assistance. That has the advantage of being easily searchable for people in the future :)

closed 2023-05-13 by nicholasbishop