rust-osdev/uefi-rs #829: Image size does not match file size?

WhyNotHugo, 2023-05-27:
I want to get a pointer to the in-memory copy of my binary, for which I'm using something that looks like this:

```rust
#[entry]
fn main(_: Handle, mut system_table: SystemTable<Boot>) -> Status {
    let current_image = boot_services.image_handle();
    let current_image = boot_services
        .open_protocol_exclusive::<LoadedImage>(current_image)
        .unwrap();

    let boot_services = system_table.boot_services();
    let (addr, size) = current_image.info();
    info!("current image is at {:?} and {:?} bytes", addr, size);

    boot_services.stall(60_000_000);
    Status::SUCCESS
}
```

This prints a size of 90112 bytes, but my compiled binary (e.g.: `target/x86_64-unknown-uefi/debug/hello-uefi.efi`) is 75264 bytes.

I've checked both my code and the implementation for `current_image.info()` and all looks good on both sides. At this point, I'm pretty sure that I'm misunderstanding something, but I'm not sure what.

How do I get a pointer to the in-memory copy of the binary? What's the cause of this discrepancy?

For context: I want to inject data into a PE section and read it at runtime.

WhyNotHugo, 2023-05-27:
Looks like using the above as a pointer points to my actual binary:

```rust
        let addr = addr as *const u8;
        let ptr = unsafe { core::slice::from_raw_parts(addr, size.try_into().unwrap()) };

        let parsed = pe_parser::pe::parse_portable_executable(ptr).unwrap();
        for section in parsed.section_table {
            info!(
                "name: {:?}, virtual_size: {:?}",
                section.get_name(),
                section.virtual_size
            );
        }
```

So I assume that there's some trailing data on the image...? Or is this just some sort of rounding that I'm not taking into consideration right now?

medhefgo, 2023-05-27:
The PE image file on disk is stored in a compact form and is not a one-to-one representation of the loaded image in memory. When a PE file is loaded, all sections are copied to their VMA/LMA, possibly zero-extending them if the final section is supposed to be larger than the data provided by the image (`BSS` sections in particular).

You can look at `objdump -x` to get an idea of the file-offset/VMA/size relationship. The size reported by the loaded image protocol *should* be the same as `SizeOfImage` from objdump.
Also, you cannot rely on the loaded image section payload to be identical to the on-disk image as relocations/fixups may have been applied them.

To get the PE data structures you can use the load address as reported by the loaded image protocol. Of course you have to make sure to use the right struct fields to access (section) data or you get the wrong stuff, ideally you use a object file crate that can read PE files that have been loaded into memory (as opposed to image files) for this.

> For context: I want to inject data into a PE section and read it at runtime.

Writing the next-gen rootkit? :D

medhefgo, 2023-05-27:
Also, the only sections you are technically allowed to modify are those that are not marked as read-only in objdump. Modifying any other can result in your app failing because of page protections (at least, that's how it's supposed to work on real hardware soon-ish I was told).

WhyNotHugo, 2023-05-27:
Ah, thanks for the explanation. So I was really missing some details here.

> To get the PE data structures you can use the load address as reported by the loaded image protocol.

So the section table in-memory is updated to match the new sizes/positions of the in-memory sections, so as long as I operate entirely on that one (and don't mix data from the in-disk one) I should be fine, right!

The sizes do match (my blob has grown in the last minutes):

- `SizeOfImage` 00021000 (0x21000 = 135168)
- size from `current_image.info()`: 135168

> Writing the next-gen rootkit? :D

No, nothing as exciting.

I'm writing a small stub loader to load a single file with a kernel+initrd+cmdline (which can be signed for SecureBoot). [Linux has deprecated the handover protocol on which gummiboot and other tools](https://gitlab.alpinelinux.org/alpine/aports/-/issues/14679) and I've been wanting to experiment in the field for some time, so this is a perfect opportunity.

I'll basically be appending the kernel, cmdline and initrd into different sections and my stub will run the kernel and pass the cmdline and initrd to it. I don't think I'd ever have dared to do anything like this without this crate :)

> Also, the only sections you are technically allowed to modify are those that are not marked as read-only in objdump. Modifying any other can result in your app failing because of page protections (at least, that's how it's supposed to work on real hardware soon-ish I was told).

I'll be modifying sections offline, not at runtime; I'll only read them at runtime, so it should be fine. I guess these protections are basically W^X. I'll keep them in mind if I ever have a fun rootkit idea.

closed 2023-05-27 by WhyNotHugo

medhefgo, 2023-05-27:
> So the section table in-memory is updated to match the new sizes/positions of the in-memory sections, so as long as I operate entirely on that one (and don't mix data from the in-disk one) I should be fine, right!

The loader copies the PE headers (including section table) as-is and does no modifications on them (technically, relocations could…). Addresses in PE header structs are RVAs (unless stated otherwise) and are relative to the load address, so to get the pointer to the section data you have to compute `load_addr + section.VirtualAddress`.

> I'm writing a small stub loader to load a single file with a kernel+initrd+cmdline (which can be signed for SecureBoot). [Linux has deprecated the handover protocol on which gummiboot and other tools relied](https://gitlab.alpinelinux.org/alpine/aports/-/issues/14679) and I've been wanting to experiment in the field for some time, so this is a perfect opportunity.

Don't tell me people are still using gummiboot? It's been superseed by sd-boot some ages ago and supports the new way of doing things.

WhyNotHugo, 2023-05-30:
Regrettably, sd-boot won't build in all the same scenarios where gummiboot worked, so it's not always a suitable replacement (see the above link as an example). Yeah, people are still using it and some of its forks.

I was mostly using it with a "there's nothing better for now" mentality, but I've always had in mind that at some point I'd just write my own stub (it's an interesting learning exercise too).