rust-osdev/uefi-rs #830: Serial Port Logger Feature Request

memN0ps, 2023-05-28:
Hi,

I want to request a serial port logger feature for the `uefi-rs` crate, please. This will enable logging through COM ports, facilitating easy debugging and analysis of UEFI applications and communication with COM ports, allowing developers to specify the port and baud rate for logging. I found a [com_logger](https://crates.io/crates/com_logger) crate for this already but I could not get it to work because it does not appear to like `uefi_services::init(&mut system_table).unwrap();` as it already initializes a logger. Unless there is a workaround for that?

medhefgo, 2023-05-28:
You can disable the logger feature on uefi-services. Though, I am pretty sure this kind of thing should be going through the serialio protocol as it'd otherwise be stepping on each other's toes.

memN0ps, 2023-05-28:
I have attempted to disable `uefi-services`, but encountered additional issues due to its dependencies on the memory allocator, logger initialization, and panic handling. Are there any relevant examples or resources you can provide to assist me with resolving these problems, please?

phip1611, 2023-05-28:
My personal opinion: just drop the uefi-services crate. Create your own logging facade, your own panic handler, and you are good to go. I never used uefi-services in any project that is bigger than a hello world :)

memN0ps, 2023-05-29:
Thank you! I got the serial port logger working but feel free to leave this open if you plan on adding it as a built-in feature.

closed 2023-05-29 by memN0ps

pathtofile, 2023-12-21:
FYI for anyone else who comes across this, if you want the panic handler from `uefi-services` but also have serial logging, you can disable the logging feature in your cargo.toml, then use the com_logger crate, e.g.:
```toml
[dependencies]
uefi-services = { version = "0.23.0", default-features = false, features = ["panic_handler"] }
com_logger = "0.1.1"
```

Tested this code using VMWare to run the .efi and read from the virtual serial port:

```rust
#![no_main]
#![no_std]

use log::*;
use uefi::prelude::*;

#[entry]
fn main(_image_handle: Handle, mut system_table: SystemTable<Boot>) -> Status {
    com_logger::builder().filter(LevelFilter::Debug).setup();
    uefi_services::init(&mut system_table).unwrap();

    debug!("Hello world!");
    info!("Hello world!");
    system_table.boot_services().stall(10_000_000);

    Status::SUCCESS
}
```