rust-osdev/uefi-rs #837: Help needed with filesystem code

DecrepitHuman, 2023-05-30:
I'm attempting to use the `FileSystem` struct through `BootServices::get_image_file_system(BootServices::image_handle())`, I've got a directory alongside my kernel image with another file inside that directory. It boots fine but then prints the following errors:
```
[TRACE] : /home/nicholas/.cargo/registry/src/index.../uefi-0.21.0/src/fs/file_system/fs.rs@327: Can't open file images: Error { status: NOT_FOUND, data: () }
[PANIC] : panicked at 'called `Result::unwrap()` on an `Err` value: Io(IoError { path: PathBuf(CString16(['i', 'm', ...])) })
```
which I'm guessing means that it's attempting to open the file inside my directory which I do not want it to do. I strictly want my kernels image to load and then continue. I would like to use this directory for other files/data that my kernel will interact with through that `FileSystem` instance. Help would be appreciated. (FYI, I have no idea how UEFI paths work so I'm trying my best here)

phip1611, 2023-05-30:
Hi! So the `.unwrap()` behind `BootServices::get_image_file_system(BootServices::image_handle())` is causing that issue?

phip1611, 2023-05-30:
> Hi, at the moment I've just got this: https://sourceb.in/FiAnPeK2QK where
> "images" is the folder I created with a blank file inside.

Please use readdir to check the file system structure that uefi sees.
Perhaps this already helps you to resolve the "not found" error

phip1611, 2023-05-30:
And this fails with not found? Interesting. You use a real uefi implementation and not ovmf, do you?

DecrepitHuman, 2023-05-30:
I'm using ovmf at the moment. My `images` folder is in the same directory as my kernels efi image.

medhefgo, 2023-05-30:
> I'm using ovmf at the moment. My `images` folder is in the same directory as my kernels efi image.

As I said, the images folder would have to be in the root folder of the file system to be found. The running EFI image's location has no bearing on this except that it has to be on the same file system as the images folder.

If in doubt provide a file tree listing of the boot image.

DecrepitHuman, 2023-05-30:
I assumes because I was getting the filesystem from the kernels image that it would default to that directory. I will fix this tomorrow and then close the issue. Thank you for the help

phip1611, 2023-05-30:
https://docs.rs/uefi/0.21.0/uefi/fs/index.html#paths

> All paths are absolute and follow the FAT-like file system conventions for paths. Thus, there is no current working directory and path components like . and .. are not supported. In other words, the current working directory is always /, i.e., the root, of the opened volume.

medhefgo, 2023-05-30:
Also, if `FileSystem` only handles stuff relative to root it would be better to refuse paths that aren't absolute to drive home that point.

phip1611, 2023-05-31:
I agree that the documentation of get_image_file_system() is indeed misleading as it doesn't directly return a protocol.

> Oh boy. That's some A-grade confusing API design.

We'll, it's documented at the most prominent possible way of the fs module. And, furthermore, when working in uefi and without an real operating system one should be relatively certainly aware that there is no such abstraction as a present working directory, tho 🤷‍♂️

The code itself has a comment somewhere I think that says present working directory support in the file system is future work. So feel free to submit a PR.

phip1611, 2023-05-31:
FileSystem::read_dir(cstr16!(r"\\"))

closed 2023-05-31 by DecrepitHuman