rust-osdev/uefi-rs #838: Trouble with user input

DecrepitHuman, 2023-05-31:
I've written a display driver using `embedded_graphics` and the `GraphicsOutput` protocol. I'm using that to display geometry & text to the screen,  and need to get keyboard input at the same time. I cannot do this at the moment because of the mutable borrow at `SystemTable<Boot>::stdin()` and the immutable borrow at `SystemTable<Boot>::boot_services()` that is used for the display driver. I can break these rules by casting the struct to a pointer and then de-referencing it to obtain a new "owned" instance but this is most definitely a ugly solution and I'm not 100% if this will cause problems (more specifically, I don't know if this will interfere with the `::boot_services()` call).

sky5454, 2024-05-11:
see exam https://github.com/slint-ui/slint/tree/master/examples/uefi-demo

phip1611, 2024-05-14:
By the way, this is going to be closed by #905

nicholasbishop, 2024-10-25:
This should be resolved by the API changes described in https://github.com/rust-osdev/uefi-rs/blob/main/docs/funcs_migration.md

closed 2024-10-25 by nicholasbishop