rust-osdev/uefi-rs #842: Core abort on Aarch64

NailAgliev, 2023-06-02:
Hello!

I`m struck with problem of getting core exception when my efi app being run.
Exception occurs when newly introduced function install_configuration_table being called. @RaitoBezarius 

I got ESR of value 0x96000004. I`m not so confident about my knowlage in armv8 arch but as i understand it correctly abort was caused by reading memory from efi runtime that mapped to efi application.

I found solution to my problem and now call is successful.

I`m new to rust so i asking for help verifying validity of my solution before creating PR.

```
install_configuration_table: unsafe extern "efiapi" fn(
        guid_entry: &Guid, <<<< i added &
        table_ptr: *const c_void
    ) -> Status,
```
```
pub unsafe fn install_configuration_table(
        &self,
        guid_entry: &Guid, <<<< i added &
        table_ptr: *const c_void,
    ) -> Result {
        (self.install_configuration_table)(guid_entry, table_ptr).to_result()
    }
```

medhefgo, 2023-06-02:
Yikes, the function sig is definitely wrong. It should indeed take a `&Guid`.

NailAgliev, 2023-06-02:
Should I create PR or this can be fixed more quickly by manteiners?

closed 2023-06-04 by nicholasbishop

NailAgliev, 2023-06-04:
Great work! Thanks for the quick follow-up. Can someone hint at when this change may appear in crates?

nicholasbishop, 2023-06-04:
Just pushed new releases to crates.io :) Thanks for the bug report!

PR #843 (2023-06-04): uefi: Fix wrong install_configuration_table() signature
Interestingly, this crashes on aarch64 but works on x86_64. Likely,
the x86_64 calling convention used still passes the Guid struct by
pointer, while the aarch64 passes it by value (or in an unexpected
registers) causing a crash.

This also adds a test case to verify this works correctly now.

Fixes: #842
commit d7697c3a (2023-06-04): uefi: Fix wrong install_configuration_table() signature

Interestingly, this crashes on aarch64 but works on x86_64. Likely,
the x86_64 calling convention used still passes the Guid struct by
pointer, while the aarch64 passes it by value (or in an unexpected
registers) causing a crash.

This also adds a test case to verify this works correctly now.

Fixes: #842