rust-osdev/uefi-rs #917: Boot Linux after running the application

JohnAZoidberg, 2023-08-11:
I'm building an application with uefi-rs and if I run it, it exits and then I try to boot Linux it crashes.
Windows works just fine.

The minimal example is:

```
#[entry]
fn main(image_handle: Handle, mut system_table: SystemTable<Boot>) -> Status {
    uefi_services::init(&mut system_table).unwrap();
    Status::NOT_FOUND
}
```

Tried on hardware and on QEMU. On QEMU it looks like this right after I select the OS from Grub

```
!!!! X64 Exception Type - 06(#UD - Invalid Opcode)  CPU Apic ID - 00000000 !!!!
RIP  - 000000003E13F018, CS  - 0000000000000038, RFLAGS - 0000000000210246
RAX  - 0000000000000000, RCX - 000000003E191E98, RDX - 0000000000000000
RBX  - 0000000000000004, RSP - 000000003FF0FD08, RBP - 000000003FF28740
RSI  - 000000003E167518, RDI - 000000003FF6A5E0
R8   - 000000003FF27140, R9  - 000000003F4807A0, R10 - 0000000000000001
R11  - 00000000000000E1, R12 - 000000003E191EE0, R13 - 0000000000000100
R14  - 000000003FF28840, R15 - 000000003E191E98
DS   - 0000000000000030, ES  - 0000000000000030, FS  - 0000000000000030
GS   - 0000000000000030, SS  - 0000000000000030
CR0  - 0000000080010033, CR2 - 0000000000000000, CR3 - 000000003FC01000
CR4  - 0000000000000668, CR8 - 0000000000000000
DR0  - 0000000000000000, DR1 - 0000000000000000, DR2 - 0000000000000000
DR3  - 0000000000000000, DR6 - 00000000FFFF0FF0, DR7 - 0000000000000400
GDTR - 000000003F9DE000 0000000000000047, LDTR - 0000000000000000
IDTR - 000000003F471018 0000000000000FFF,   TR - 0000000000000000
FXSAVE_STATE - 000000003FF0F960
!!!! Find image based on IP(0x3E13F018) (No PDB)  (ImageBase=000000003DF1E000, EntryPoint=000000003DF1F000) !!!!
```

JohnAZoidberg, 2023-08-11:
Root cause is the `exit_boot_services` hook in `init`:

```
        boot_services
            .create_event(
                EventType::SIGNAL_EXIT_BOOT_SERVICES,
                Tpl::NOTIFY,
                Some(exit_boot_services),
                None,
            )
            .map(|_| ())
```

Even if I define it as 

```
unsafe extern "efiapi" fn exit_boot_services(_e: Event, _ctx: Option<NonNull<c_void>>) {
}
```

I see the same crash. The only thing that helps is to remove the call to `create_event`.

nicholasbishop, 2023-08-12:
Thanks for filing this. I put together a small test that reproduces your crash: https://github.com/nicholasbishop/uefi-rs/tree/bishop-quick-exit

The problem is that `uefi-services` is geared towards bootloaders, where it's expected that the application won't exit. E.g. if a bootloader app uses `uefi-services` and launches the Linux kernel, early in startup Linux will call `ExitBootServices` and the bootloader app will still be in memory and its event handler will be able to run without issue.

In the case you've outlined, the application uses `uefi-services` to register an event handler, and that handler never gets unregistered when the application exits, and so exiting boot services triggers a crash.

I'm not sure yet how we want to solve this. One option would be to change the API so that the caller does something like `let _app = uefi_services::init();`, and that returned object has a `Drop` impl that removes the event handler. That might be to easy to get wrong on the caller side though, by accidentally dropping it too early.

JohnAZoidberg, 2023-08-15:
Ohh that makes sense.

Perhaps we could force the user to return object from main and the wrapper drops it.

But actually I noticed this while I'm trying to register my own callback that must run just before the OS boots, not when the app exits.

JohnAZoidberg, 2023-08-15:
I modified `uefi-services` to return the event:

```diff
diff --git i/uefi-services/src/lib.rs w/uefi-services/src/lib.rs
index 0b715b1..ee1fd04 100644
--- i/uefi-services/src/lib.rs
+++ w/uefi-services/src/lib.rs
@@ -78,11 +78,11 @@ pub fn system_table() -> NonNull<SystemTable<Boot>> {
 ///
 /// This must be called as early as possible,
 /// before trying to use logging or memory allocation capabilities.
-pub fn init(st: &mut SystemTable<Boot>) -> Result {
+pub fn init(st: &mut SystemTable<Boot>) -> Result<Option<Event>> {
     unsafe {
         // Avoid double initialization.
         if SYSTEM_TABLE.is_some() {
-            return Status::SUCCESS.to_result();
+            return Status::SUCCESS.to_result_with_val(|| { None });
         }
 
         // Setup the system table singleton
@@ -104,7 +104,7 @@ pub fn init(st: &mut SystemTable<Boot>) -> Result {
                 Some(exit_boot_services),
                 None,
             )
-            .map(|_| ())
+            .map(|e| Some(e))
     }
 }
```

Then in the app I can run:

```rs
#[entry]
fn main(image_handle: Handle, mut system_table: SystemTable<Boot>) -> Status {
    let event: Option<Event> = uefi_services::init(&mut system_table).unwrap();
    let boot_services = system_table.boot_services();

    println!("Hello World");

    if let Some(event) = event {
        // Will creash when Linux boots if we don't call this
        boot_services.close_event(event).unwrap();
    }
    Status::SUCCESS
}
```

and it will boot to the Linux just fine.

nicholasbishop, 2023-10-25:
I've merged https://github.com/rust-osdev/uefi-rs/pull/920 for this, and the remaining work is tracked in #943.

closed 2023-10-25 by nicholasbishop