rust-osdev/uefi-rs #921: How to obtain a protocol interface

scholzp, 2023-08-17:
Hi all,
in my current use case I try to obtain a protocol interface pointer, not for the purpose of using it, but rather to uninstall the protocol from an handle. This is necessary as one is required to call [`Bootservices::uninstall_protocol_interface`](https://docs.rs/uefi/latest/uefi/table/boot/struct.BootServices.html#method.uninstall_protocol_interface). Normally for such functionality I would use `EFI_BOOT_SERVICES.HandleProtocol()` (see spec 7.3.7), which would return the interface pointer. This function is not implemented in the crate and [`Bootservices::test_protocol`](https://docs.rs/uefi/latest/uefi/table/boot/struct.BootServices.html#method.test_protocol), which would conceptually achieve the same result as `HandleProtocol`, doesn't return a pointer to the interface either. 

For me the only possible way to do this seems to be by using [open_protocol](https://docs.rs/uefi/latest/uefi/table/boot/struct.BootServices.html#method.open_protocol), dereferenceing the `ScopedProtocol`, somehow create a pointer from it and then use this to uninstall the protocol, after I closed it of course. This feels a bit hacky to me and I would prefer a more intuitive way for obtaining an interface pointer. Is there any better way to do this, of which I'm unaware? If not, would you be interested in such a feature?

Thanks for your help!

nicholasbishop, 2023-09-02:
I had to ponder this a bit :) It is possible to do with the current API, and I've put up a PR to demonstrate this in the test-runner: https://github.com/rust-osdev/uefi-rs/pull/931

It's not exactly the most intuitive thing though, so I'm open to suggestions on improving this. I expect uninstalling a protocol interface to be relatively uncommon though, so maybe it's OK for it to be a little clunky -- I'm unsure.

scholzp, 2023-09-04:
Hi Nicholas, thank you for your response and the example! :)

I think the way you demonstrated this is the way, it should be done when using the crate. I've implemented a `handle_protocol` wrapper in the meantime and didn't find a way to really integrate it in a useful way into this crate. It would introduces a lot of unsafety in my option when handling protocols, as the handler isn't automatically registered by the UEFI as it would be the case when using `open_protocol`. So it might be better if there only exists one way of obtaining a interface pointer. I too think, that uninstalling a protocol interface should be a uncommon task and in the seldom cases where it's useful, it's OK to be a bit clunky. But this is just my thoughts about this.

So again, thank you for your answer! :)

phip1611, 2025-04-09:
Closing, fixed by #931

closed 2025-04-09 by phip1611