rust-osdev/uefi-rs #942: Potentially Outdated QEMU command in book

0x32c2cac3info, 2023-09-16:
TL;DR [Launch the VM](https://github.com/rust-osdev/uefi-rs/blob/main/book/src/tutorial/vm.md#launch-the-vm) might have an outdated qemu-system command. I got mine working via:

```sh
qemu-system-x86_64 -enable-kvm -drive if=pflash,format=raw,file=OVMF_CODE.fd -drive if=pflash,format=raw,file=OVMF_VARS.fd -drive format=raw,file=fat:rw:esp
```

There were some knits leading up to this (on gentoo) regarding some libslirp.so dependencies. I can potentially help others resolve this if they're experiencing similar issues.

Back on topic though - it seems the necessary command has readonly-on=on required due to a deprecation, and that readonly is no longer allowed on raw (I guess?)

nicholasbishop, 2023-09-16:
I tested on a couple versions of qemu and don't see any error, so need some additional info. Can you post the full output of the failing command? Also, what does `qemu-system-x86_64 --version` show?

nicholasbishop, 2023-10-24:
Closing for now -- feel free to reopen if you can repro the error.

closed 2023-10-24 by nicholasbishop