rust-osdev/uefi-rs #954: Release uefi-0.24.0

phip1611, 2023-10-06:
It's quite some time since the last release. We releases `0.24` in June and we should Release `0.25` next. (I know someone that asked politely for a new release.)

PS: @nicholasbishop If you don't mind, maybe I can perform the release this time? You always did this so far. Might be good if I'm also familiar with the process so that we don't have a single-point-of-failure :) Anything I need to now beside whats written in [PUBLISHING.md](https://github.com/rust-osdev/uefi-rs/blob/main/PUBLISHING.md)?

phip1611, 2023-10-06:
Is there a reason why there aren't any entries for `uefi-raw` in the changelog?

nicholasbishop, 2023-10-06:
> Is there a reason why there aren't any entries for uefi-raw in the changelog?

Not a strong one, I just figured that since the library was still in the early stages and churning a lot, changelogs wouldn't be very useful. But we could start adding them going forward.

> If you don't mind, maybe I can do the release this time? You always did this so far. Might be good if I'm also familiar with the process so that we don't have a single-point-of-failure :) Is there anything I need to know for that beside what's in [PUBLISHING.md](https://github.com/rust-osdev/uefi-rs/blob/main/PUBLISHING.md)?

Sounds good! I think PUBLISHING.md is accurate, but I've been following some local notes I made that use cargo-release. There's one surprising thing, which is that cargo release isn't updating version numbers automatically, and I don't know why. It works in other projects, so must be something weird in our repo. I've been meaning to integrate these notes into publishing.md, but haven't gotten around to it yet. Here's what I have in my notes:

1. File a release bug. See <https://github.com/rust-osdev/uefi-rs/issues/637> for example. Doesn't need to contain much, just the versions you intend to release.
2. Make sure everything you intend to have merged in is merged.
3. Install or update the [cargo-release] tool: `cargo install cargo-release`
4. Make sure your local repo is up to date: `git checkout main && git pull`
5. Checkout a new branch to contain the release commits: `git checkout -b <some-branch-name>`
6. Push the branch to the main repo (not a fork): `git push -u origin <some-branch-name>`
7. Run `cargo release` in dry-run mode. For example: `cargo release -p uefi-raw 0.1.0`
8. Run it for real: `cargo release -p uefi-raw 0.1.0 --execute`
9. Update `uefi-raw` version in `uefi/Cargo.toml`. Make sure to `cargo xtask build` after to update the lock file. Commit the change. TODO: why isn't cargo release handling this?
10. Order of releases: `uefi-raw`, `uefi-macros`, `uefi`, `uefi-services`
11. Update deps in `template/Cargo.toml` and `book/src/tutorial/app.md`
12. Update changelog.
13. Merge branch into main via PR.
14. Close ticket.

[cargo-release]: https://github.com/crate-ci/cargo-release

phip1611, 2023-10-09:
Interesting, I've never heard of [`cargo-release`](https://crates.io/crates/cargo-release) , I always did the relevant steps manually in my own projects.

phip1611, 2023-10-09:
Closing in favor of #955

closed 2023-10-09 by phip1611

phip1611, 2023-10-09:
Hm, on a first glance, getting to know how `cargo release` works takes 10x more time than doing things manually. I hope I don't mess something up

phip1611, 2023-10-09:
@nicholasbishop you are the owner of `uefi-raw`. Can you please add @GabrielMajeri  and me?

nicholasbishop, 2023-10-09:
Ah sorry about that! I have sent you both invites.