rust-osdev/uefi-rs #970: Rething (im)mutabiltiy of device paths (`DevicePathBuilder` / `install_protocol_interface`)

phip1611, 2023-10-23:
`DevicePathBuilder::finalize()` returns a `&DevicePath`. If one wants to install this to a handle using `install_protocol_interface`, one has to cast that to a `*mut c_void`. This feels unhandy.

For example, this is done here: https://github.com/nix-community/lanzaboote/pull/194

phip1611, 2023-10-24:
🔝 I wonder if this is possible without (procedural) macros but only const functions 🤔

nicholasbishop, 2023-10-29:
I agree with you that install_protocol_interface could just take a const pointer. Put up https://github.com/rust-osdev/uefi-rs/pull/981 for that.

Interesting idea about constructing const device paths. It's tricky in the general case since some nodes are rather dynamic. We'd also have to change our current trait-based approach (https://docs.rs/uefi/latest/uefi/proto/device_path/build/trait.BuildNode.html), since I don't think there's any way to declare const fns in traits yet.

nicholasbishop, 2023-10-30:
I opened https://github.com/rust-osdev/uefi-rs/issues/983 for building device paths as constants. `install_protocol_interface` & co have been updated to take a const pointer, so closing this issue.

closed 2023-10-30 by nicholasbishop