rust-osdev/uefi-rs #1037: How to make a runtime driver? 🤔

NiiightmareXD, 2023-12-25:
I'm trying to create a UEFI driver that hooks get_variable before the actual OS starts, but I'm running into some problems. Firstly, when I try to load the EDK II shell, it says that it's not a driver. Secondly, when I run it from BIOS, everything freezes after it's done. I think this is because the UEFI phase has ended. 

Can you please advise me on how to make a driver? 

I apologize if this issue might not be related to uefi-rs, and if I did something wrong. I'm relatively new to osdev. 

Thank you for your help.

NiiightmareXD, 2023-12-27:
the freezing was because the hook was null but I still don't know about edk II driver

nicholasbishop, 2023-12-27:
Hi, can you link to a repository with your code?

NiiightmareXD, 2023-12-28:
> Hi, can you link to a repository with your code?

Hi, I don't have a repository for it yet and I don't think my code is the problem because it doesn't even run when I use `load myproj.efi` in edk 2 shell it just refuses to load it and says not a driver.

But I can upload my project if needed.

nicholasbishop, 2023-12-28:
I can't say for sure without seeing the code and how you are building it, but my first guess would be that you are using one of Rust's default UEFI target specifications. These inherit from [`uefi_msvc.rs`](https://github.com/rust-lang/rust/blob/f4d794ea0b845413344621d89f6c945062748485/compiler/rustc_target/src/spec/base/uefi_msvc.rs#L33), which sets the `/subsystem:efi_application` linker flag. For a runtime driver you want `efi_runtime_driver` instead.

You can change this by setting `link-args`:
```
# In .cargo/config.toml:
[build]
rustflags = ["-C", "link-args=/subsystem:efi_runtime_driver"]
```

NiiightmareXD, 2023-12-28:
Yes, that was the problem thanks for helping.

closed 2023-12-28 by NiiightmareXD

phip1611, 2023-12-31:
Do we have this in the docs somewhere? I think, we should add it to the readme. It's good to have it there.

NiiightmareXD, 2024-01-08:
I think it should be in the book here https://rust-osdev.github.io/uefi-rs/HEAD/tutorial/app.html

nicholasbishop, 2024-01-15:
Added here: https://rust-osdev.github.io/uefi-rs/HEAD/how_to/building_drivers.html

I also added the same example to https://doc.rust-lang.org/rustc/platform-support/unknown-uefi.html, but it hasn't propagated there yet.

phip1611, 2024-01-15:
Amazing, thanks Nicholas!