rust-osdev/uefi-rs #1084: how can I found the `Miscellaneous Protocols` and use it

sky5454, 2024-02-28:
I want to get the timestamp in uefi, and the docs [39. Miscellaneous Protocols](https://uefi.org/specs/UEFI/2.10/39_Micellaneous_Protocols.html?highlight=timestamp#) in UEFI Spec 2.10.

the docs' context as follow:
[image]

but I couldn't found and misc protocols or any timestamp api in `uefi-rs`, could you help me for giving some suggestion. thanks!

nicholasbishop, 2024-03-01:
Hi, that protocol is not currently implemented in uefi-rs. (Contributions welcome, if you are interested in implementing it!) Depending on what your use case is, https://docs.rs/uefi/latest/uefi/table/runtime/struct.RuntimeServices.html#method.get_time might be an acceptable alternative.

sky5454, 2024-03-06:
> Hi, that protocol is not currently implemented in uefi-rs. (Contributions welcome, if you are interested in implementing it!) Depending on what your use case is, https://docs.rs/uefi/latest/uefi/table/runtime/struct.RuntimeServices.html#method.get_time might be an acceptable alternative.

if I want to implement it, how could I found the `efiapi` which should be used. Is there any references or docs?

nicholasbishop, 2024-03-08:
First add the raw protocol in this directory: https://github.com/rust-osdev/uefi-rs/tree/main/uefi-raw/src/protocol. This could go in a new `misc.rs` module. See [`protocol/loaded_image.rs`](https://github.com/rust-osdev/uefi-rs/blob/main/uefi-raw/src/protocol/loaded_image.rs) for an example of what this would look like -- basically just adding a `TimestampProtocol` struct containing two function pointers, matching what's in the [spec](https://uefi.org/specs/UEFI/2.10/39_Micellaneous_Protocols.html), and an associated `GUID` constant.

Then you'll add the higher-level version to the `uefi` package, in this directory: https://github.com/rust-osdev/uefi-rs/tree/main/uefi/src/proto. Again this would be a `misc.rs` module, see [`proto/loaded_image.rs`](https://github.com/rust-osdev/uefi-rs/blob/main/uefi/src/proto/loaded_image.rs#L17) for an example. Add a `repr(transparent)` `Timestamp` struct, and implement `get_timestamp` and `get_properties`.

That's the general outline, happy to answer questions about any of the details. And/or, feel free to put up a PR even if it's not complete, and I can make comments there too.

sky5454, 2024-03-11:
> First add the raw protocol in this directory: https://github.com/rust-osdev/uefi-rs/tree/main/uefi-raw/src/protocol. This could go in a new `misc.rs` module. See [...]

like this:
https://github.com/rust-osdev/uefi-rs/blob/86e200e277323937187771342b84e8f6e6efc562/uefi-raw/src/protocol/console/serial.rs#L78-L98
but I didn't know how to found out the `"efiapi"` 's `set_control_bits` func definition.

nicholasbishop, 2024-03-11:
Using [`EFI_SERIAL_IO_PROTOCOL`](https://uefi.org/specs/UEFI/2.10/12_Protocols_Console_Support.html#efi-serial-io-protocol) as an example, it's defined in the spec as:
```c
typedef struct {
 UINT32                       Revision;
 EFI_SERIAL_RESET             Reset;
 EFI_SERIAL_SET_ATTRIBUTES    SetAttributes;
 EFI_SERIAL_SET_CONTROL_BITS  SetControl;
 EFI_SERIAL_GET_CONTROL_BITS  GetControl;
 EFI_SERIAL_WRITE             Write;
 EFI_SERIAL_READ              Read;
 SERIAL_IO_MODE               *Mode;
 CONST EFI_GUID               *DeviceTypeGuid; // Revision 1.1
} EFI_SERIAL_IO_PROTOCOL;
```

The type of `SetControl` is `EFI_SERIAL_SET_CONTROL_BITS`, which is a typedef defined under [`EFI_SERIAL_IO_PROTOCOL.SetControl`](https://uefi.org/specs/UEFI/2.10/12_Protocols_Console_Support.html#efi-serial-io-protocol-setcontrol):

```c
typedef
EFI_STATUS
(EFIAPI *EFI_SERIAL_SET_CONTROL_BITS) (
 IN EFI_SERIAL_IO_PROTOCOL                   *This,
 IN UINT32                                   Control
 );
```

You can usually translate these function types pretty directly in Rust. Note that we don't bother with creating a `type` alias in the Rust API, you can just write the function pointer type inline in the struct definition.

One last note, our definition of the function does look a little different from C in that the second arg's type is `ControlBits` instead of a `u32`, but that `ControlBits` type is itself a wrapper around a `u32` type.

nicholasbishop, 2024-03-30:
I've put up a PR to add support for this protocol: https://github.com/rust-osdev/uefi-rs/pull/1109

Unfortunately, despite this having been added to the UEFI spec a while ago, it doesn't seem to be well supported. I tested a recent Lenovo, as well as OVMF in QEMU, and neither one has this protocol.

sky5454, 2024-03-31:
Weee!! I just completed the support of TimestampProtocol today yeah.
And another things, I have a doubt about `ResetNotificationProtocol` and didn't know how to code it, it has a pointer to func which confuses me.

Docs: 
 - https://uefi.org/specs/UEFI/2.10/39_Micellaneous_Protocols.html?highlight=timestamp#reset-notification-protocol

EDK2: 

https://github.com/tianocore/edk2/blob/37f63deeefa89b7ac28285bc59eca22a5c00eb92/MdePkg/Include/Protocol/ResetNotification.h#L21-L48

https://github.com/tianocore/edk2/blob/37f63deeefa89b7ac28285bc59eca22a5c00eb92/MdePkg/Include/Uefi/UefiSpec.h#L1072-L1095

[image]

sky5454, 2024-03-31:
> I've put up a PR to add support for this protocol: #1109
> 
> Unfortunately, despite this having been added to the UEFI spec a while ago, it doesn't seem to be well supported. I tested a recent Lenovo, as well as OVMF in QEMU, and neither one has this protocol.

I think its implementation is related to the UEFI Spec version number commonly used by vendors, probably an ancient version. It seem that I should find another way to get timestamp.

nicholasbishop, 2024-03-31:
One option you could try (if you are on x86) is the [`RDTSC`](https://doc.rust-lang.org/core/arch/x86_64/fn._rdtsc.html) instruction, or [`RDTSCP`](https://doc.rust-lang.org/core/arch/x86_64/fn.__rdtscp.html). These can be used to read a timestamp counter.

Depending on exactly what your needs are, this can be somewhat complex; see [How to Benchmark Code Execution
Times on Intel IA-32 and IA-64 Instruction Set Architectures](https://cis.temple.edu/~qzeng/cis3207-spring18/files/ia-32-ia-64-benchmark-code-execution-paper.pdf) for details of execution barriers. Whether those details matter probably depends on whether you are microbenchmarking or looking at something longer running.

There's also the question of units. On new-ish processors the timestamp frequency is invariant, and on sufficiently new processors you can directly query the scaling factors needed to convert timestamps to seconds. See for example [this code in edk2](https://github.com/tianocore/edk2/blob/37f63deeefa89b7ac28285bc59eca22a5c00eb92/UefiCpuPkg/Library/CpuTimerLib/CpuTimerLib.c#L42). On older processors you might need to figure out the frequency by measuring the timestamp on both sides of a call to `BootServices::stall`. On the other hand, you can skip all of that if you only care about relative duration and don't need to convert to seconds.

nicholasbishop, 2024-03-31:
> And another things, I have a doubt about ResetNotificationProtocol and didn't know how to code it, it has a pointer to func which confuses me.

A function pointer in Rust looks very much like a regular function definition. Here's an example: https://play.rust-lang.org/?version=stable&mode=debug&edition=2021&gist=b8ce6ab497a1119b2db8e1a22ac98742
```rust
extern "C" fn f2() {}

fn f1(fn_ptr: extern "C" fn() -> ()) {
    dbg!(fn_ptr);
}

fn main() {
    f1(f2);
}
```

sky5454, 2024-04-01:
> Reference

It seems nothing different with C. 
Actually, I was going to ask you if some design about the callback, I am new ruster, and for other language like Golang/Java has lambda function design. rust also has something like 
```rust

let func_ptr = |x| -> { 
  fmt.println(x);
  unsafe{ 
      /* some code for ABI call */
  }
}

set_callback(type, func_ptr);

```

I wanna know if uefi-rs has any design or paradigm? I don't wanna break the rules. 
Thanks for your guidance!

sky5454, 2024-04-01:
> One option you could try (if you are on x86) is the [`RDTSC`](https://doc.rust-lang.org/core/arch/x86_64/fn._rdtsc.html) instruction, or [...]

yeah, the one that I need now is only the duration counter. I am an embeded developer for Arm Soc, it has a timer/counter which meet demand, And x86 should also has it. But now I try develop uefi app/driver at x86, just for fun, coding fun! but x86 is really complex which I couldn't easy to get the way.

<br/><br/><br/><br/>

thank you, I read the code, found they are using the reg named TSC (Time Stamp Counter), though affected by cpu hz. I ever use it in Linux.

and also could use the the [ACPI Timer](https://wiki.osdev.org/ACPI_Timer), [HPET, or High Precision Event Timer](https://wiki.osdev.org/HPET), [AP IC Timer](https://wiki.osdev.org/APIC_timer)/PIC and so on. see: 

I found blog on the internet which recommend `constant TSC > HPET > ACPI_PM > TSC > i8254` in Linux.

nicholasbishop, 2024-04-05:
> Actually, the things I wanna ask you if some design about the callback,

Regarding how to use an implement callbacks, a good example is BootService::create_event:
https://github.com/rust-osdev/uefi-rs/blob/6093205c3eb27b2e78be4c003c04d46679bff420/uefi/src/table/boot.rs#L317

And here's an example of calling it:
https://github.com/rust-osdev/uefi-rs/blob/6093205c3eb27b2e78be4c003c04d46679bff420/uefi-test-runner/src/boot/misc.rs#L40

Note that although Rust does have closures (`|param| { body }`, those can't be used here directly because we need to pass in a function with the correct ABI (`efiapi`).

sky5454, 2024-04-05:
PR #1116 , though in OVMF run at `cargo run xtask`, review it please, because it print warn log as wo code, hope that I don't make some mistake:
```sh
[ INFO]: uefi-test-runner\src\proto\misc.rs@020: Running loaded Timestamp Protocol test
[ WARN]: uefi-test-runner\src\proto\misc.rs@034: Failed to open Timestamp Protocol: Error { status: UNSUPPORTED, data: () }
[ INFO]: uefi-test-runner\src\proto\misc.rs@041: Running loaded ResetNotification protocol test
[ WARN]: uefi-test-runner\src\proto\misc.rs@076: Failed to open ResetNotification Protocol: Error { status: UNSUPPORTED, data: () }

```

phip1611, 2024-04-05:
Well, OVMF doesn't has the protocol compiled in, I suppose? Is there an easy way to add it to our pre-build OVMF version, @nicholasbishop ?

sky5454, 2024-04-05:
> Well, OVMF doesn't has the protocol compiled in, I suppose? Is there an easy way to add it to our pre-build OVMF version, @nicholasbishop ?

I run `target/x86_64-unknown-uefi/debug/uefi-test-runner.efi` in VMware, and uefi-test-runner has a assert like this：

[image]
[image]

https://github.com/rust-osdev/uefi-rs/blob/6093205c3eb27b2e78be4c003c04d46679bff420/uefi-test-runner/src/main.rs#L24-L42

I think it should be replaced as warnning.

nicholasbishop, 2024-04-07:
Regarding the timestamp protocol, it seems to me that for whatever reason edk2 doesn't broadly enable it. I'm not 100% sure, but I don't think there's an easy way to just enable it at compile time; some code changes would be needed.

For the uefi-test-runner, it's intentional that it only works under QEMU. Originally it was more flexible and would only warn instead of fail if the environment was different, but this made the test less useful for actually finding errors. Several times I introduced a change that accidentally broke something, but we didn't notice because the test runner still "passed". So, we changed it to require QEMU, and provided prebuilt OVMF files that have various extra things enabled. See https://github.com/rust-osdev/uefi-rs/issues/553 for some more info.

closed 2024-04-20 by nicholasbishop