rust-osdev/uefi-rs #1131: Feature Request: Implementing MP Protocol Compatibility in Hyper-V

memN0ps, 2024-04-15:
I've discovered that Hyper-V does not currently support the MP (Multi-Processor) protocol. This limitation prevents the use of MP protocol functionalities within Hyper-V environments. The absence of this support is evident in the `processor.rs` file from the [illusion-rs project lines 38-40](https://github.com/memN0ps/illusion-rs/blob/development/uefi/src/processor.rs#L38-L40). Could support for the MP protocol be implemented in future updates to facilitate its use please?

[image]
**Figure 1: Loading Attempt of `illusion.efi`**

[image]
**Figure 2: Serial Output Logs Documenting the Failure**

Thanks.

nicholasbishop, 2024-04-15:
Is it failing on `let handle = boot_services.get_handle_for_protocol::<MpServices>()?;`? If so, I don't think we can do anything about that, as it indicates the UEFI environment (provided by the hypervisor, not us) is lacking this protocol.

memN0ps, 2024-04-15:
Thank you for getting back to me. Indeed, the failure occurs at `let handle = boot_services.get_handle_for_protocol::<MpServices>()?;`, and I understand that this reflects a limitation of the UEFI implementation provided by Hyper-V, not in the crate.

Given this, it seems the best path forward would be to utilize ACPI instead of the MP protocol for managing multi-processor setups. Could we explore adding ACPI support within the project to handle such configurations? This would provide a more flexible and modern approach to processor management in environments where MP protocol support is unavailable.

I should have been clearer in my initial inquiry about aiming to use ACPI for this purpose. Looking forward to your thoughts on how we might integrate this.

phip1611, 2024-04-15:
> Given this, it seems the best path forward would be to utilize ACPI instead of the MP protocol for managing multi-processor setups. Could we explore adding ACPI support within the project to handle [...]

This is the scope of the operating system. The `uefi` crate is as much as possible unopinionated about how you write an OS-specific loader or kernel. These are implemenentation details.

So, write your own OS-specific loader or kernel as EFI app, exit the boot services, gain control over the hardware, and awake the other APs. An example how other APs are activated using IPIs can be seen in the Hedron microkernel: https://github.com/cyberus-technology/hedron/blob/8ab6bd0bd6468be4602f65280a58ced1b1999156/src/lapic.cpp#L98

closed 2024-04-15 by memN0ps