rust-osdev/uefi-rs #1137: Whether a lifecycle container/manager can be provided to store the currently open protocol?

sky5454, 2024-04-15:
Whether a lifecycle container/manager can be provided to store the currently open protocol? 
https://docs.rs/uefi/latest/uefi/table/boot/enum.OpenProtocolAttributes.html
[image]

Will you design a object manager which stores opened protocol by `Exclusive `, then user no need to use the unsafe `GetProtocol`.

https://github.com/rust-osdev/uefi-rs/issues/893

sky5454, 2024-04-15:
```rust
static mut MYPROTOCOL_RUNNING_ID = 0;

fn efI_main() {
   myprotocol = open_protocol_exclusive();
   MYPROTOCOL_RUNNING_ID = store_opened_protocol_to_hashmap(myprotocol);
}

fn in_other_func() {
  myprotocol  = get_opened_protocol_by_hashmap(MYPROTOCOL_RUNNING_ID)
                  .execpt("The previously opened protocol has been closed");

}
```

nicholasbishop, 2024-04-15:
I may be misunderstanding you, but I think https://docs.rs/uefi/latest/uefi/table/boot/struct.BootServices.html#method.open_protocol_exclusive and https://docs.rs/uefi/latest/uefi/table/boot/struct.ScopedProtocol.html already handle this. If you call `open_protocol_exclusive`, you'll get an exclusive and scoped protocol that will be closed on drop. If you tried to call `open_protocol_exclusive` on that same protocol and handle while it was already opened exclusively, you'd get an error.

sky5454, 2024-04-16:
but the complex rust structure with static pass in stack is inconvenient. and it needs box wrapped.  and no Copy trait.

nicholasbishop, 2024-04-17:
https://github.com/rust-osdev/uefi-rs/issues/893 will help with that to some extent, since we'll be able to drop the `'boot` lifetime parameter from things like `ScopedProtocol`. I don't think we'd ever want to support `Copy` though, I don't see how that would work with an exclusive protocol?

nicholasbishop, 2025-02-09:
Closing as I don't think we currently have anything actionable to do here; please feel free to reopen or file a new issue as needed.

closed 2025-02-09 by nicholasbishop