rust-osdev/uefi-rs #1138: Example application fails to build with "could not find `helpers` in `uefi`"

keli5, 2024-04-17:
hello! I'm trying to build the example application and I seem to have an issue that I can't find any documentation on. I've followed the [2.1 Creating a UEFI Application](https://rust-osdev.github.io/uefi-rs/HEAD/tutorial/app.html) section to the letter, but I get the below error when trying to build it:

```
letters@DESKTOP-RGK8TVG:~/src/rsfi$ cargo build --target x86_64-unknown-uefi
   Compiling bitflags v2.5.0
   Compiling ptr_meta v0.2.0
   Compiling uguid v2.2.0
   Compiling bit_field v0.10.2
   Compiling uefi-raw v0.5.1
   Compiling ucs2 v0.3.3
   Compiling log v0.4.21
   Compiling cfg-if v1.0.0
   Compiling uefi v0.27.0
   Compiling uefi-services v0.24.0
   Compiling rsfi v0.1.0 (/home/letters/src/rsfi)
error[E0433]: failed to resolve: could not find `helpers` in `uefi`
 --> src/main.rs:9:11
  |
9 |     uefi::helpers::init(&mut system_table).unwrap();
  |           ^^^^^^^ could not find `helpers` in `uefi`

For more information about this error, try `rustc --explain E0433`.
error: could not compile `rsfi` (bin "rsfi") due to 1 previous error
```

I'm using Debian on WSL2 if that makes any difference. My main.rs, rust-toolchain.toml, etc. all match the example program to the letter as far as I can tell. Any help is appreciated, thanks!

nicholasbishop, 2024-04-18:
This is due to a recent API change on the `main` branch. It's not yet released, but the book is built from the `main` rather than the latest release. To fix it, change that line to `uefi_services::init(&mut system_table).unwrap();`, and add `uefi-services` as a dep in `Cargo.toml`.

See https://github.com/rust-osdev/uefi-rs/tree/eaa146fbc02195a9d6846caeb033b55159a9da9b/template for an example.

Sorry about the confusion here, we'll do a new release soon to resolve this mismatch.

keli5, 2024-04-18:
Fixed, thank you!

closed 2024-04-18 by keli5