rust-osdev/uefi-rs #1139: `uefi::helpers` not found / Rust UEFI Book out of sync

osteffenrh, 2024-04-19:
The UEFI app tutorial at https://rust-osdev.github.io/uefi-rs/HEAD/tutorial/app.html does not work with the current version of the uefi crate available from crates.io.
The book refers to the latest state of the source repo, and not what is published on creates.io, although it instructs to use `cargo add uefi` to add the crate.

The current crates.io version (0.27.0) does not have the `uefi::helpers` mod yet, and thus stuff does not build -- very confusing.

Maybe the Rust book needs to be versioned?
Or only rebuild it when the create is published to crates.io?

Yes, the README.md at the uefi-v0.27.0 tag points to a working example, but the [documentation linked at crates.io](https://docs.rs/uefi/0.27.0/uefi/) also points to the Rust book, following HEAD instead of pointing to the correct version.

phip1611, 2024-04-19:
Thanks for the report.

It's a duplicate of https://github.com/rust-osdev/uefi-rs/issues/1138.

We made that API change very recently. We should publish the new crate versions ASAP, @nicholasbishop . Then these kinds of problems will no longer occur.

closed 2024-04-19 by phip1611

phip1611, 2024-04-19:
See #1140 - soon the problem will be gone :)