rust-osdev/uefi-rs #1169: No graphics

l420y, 2024-05-15:
Hi. Developing on Fedora Linux. I'm trying to run an example (https://rust-osdev.github.io/uefi-rs/HEAD/how_to/drawing.html) with VM launch from book (https://rust-osdev.github.io/uefi-rs/HEAD/tutorial/vm.html), VM starts, cannot boot and closes after 5 seconds. But if I run a uefi application without graphics everything works correctly, info!() macro output string. I debugged and found that application stops responding at the stage of creating a GOP from the GOP handle and the VM closes (`let mut gop = bt.open_protocol_exclusive::<GraphicsOutput>(gop_handle)?;`). What should I do? 

**sry  for the stupid questions** 

qemu launch:
`qemu-system-x86_64 -enable-kvm \
    -drive if=pflash,format=raw,readonly=on,file=OVMF_CODE.fd \
    -drive if=pflash,format=raw,readonly=on,file=OVMF_VARS.fd \
    -drive format=raw,file=fat:rw:esp
`

osteffenrh, 2024-05-16:
I can reproduce this.

Running the example from the uefi-rs repo with `cargo xtask run --example sierpinski
` works though.
Going through the command line options used there, I found that `-device virtio-rng-pci` is needed.

l420y, 2024-05-16:
Thank u, everything works  ❤️

closed 2024-05-16 by l420y