rust-osdev/uefi-rs #1204: make MemoryMap own its buffer (Vec)

DaniD3v, 2024-06-13:
To save your system Memory Map you need to call `boot_services.memory_map_size()` and allocate. I don't think doing this is possible without heap allocations? If that's the case why not just make MemoryMap own a vector, make memory_map_size() private (to increase usability) and remove the lifetime on MemoryMap?

I don't think there would be any downsides to doing that. I don't have a lot of experience in either UEFI or OSdev so feel free to correct me if I'm wrong

nicholasbishop, 2024-06-13:
You're in luck, there's a PR in progress to do exactly that: https://github.com/rust-osdev/uefi-rs/pull/1175

phip1611, 2024-06-13:
Yes! I hope I can continue working on it in the next couple of days