rust-osdev/uefi-rs #1236: Need help for ACPI table, rsdt and xsdt entries are aberrant

Zekirax, 2024-07-09:
I want to print all acpi headers. I can detect the rsdt and xsdt tables and their checksums are correct. But the entries in xsdt all have a length of 0 and those in rsdt a length of more than 1 billion.  All the signatures are not ascii characters.

my code

```rs
#![no_main]
#![no_std]

use core::str::from_utf8_unchecked;

use log::info;
use uefi::{prelude::*, table::cfg::ACPI2_GUID};

#[repr(C, packed)]
#[derive(Debug)]
struct RSDP {
    signature: [u8; 8],
    checksum: u8,
    oem_id: [u8; 6],
    revision: u8,
    rsdt_address: u32,
    length: u32,
    xsdt_address: u64,
    extended_checksum: u8,
    reserved: [u8; 3],
}
/// acpi spec 5.2.7
#[repr(C, packed)]
#[derive(Debug, Clone, Copy)]
struct ACPISDTHeader {
    signature: [u8; 4],
    length: u32,
    revision: u8,
    checksum: u8,
    oem_id: [u8; 6],
    oem_table_id: [u8; 8],
    oem_revision: u32,
    creator_id: u32,
    creator_revision: u32,
}
impl ACPISDTHeader {
    #[inline]
    fn get_entries_len(&self, entry_ptr_size: usize) -> usize {
        let entries_size = self.length as usize - size_of::<Self>();
        entries_size / entry_ptr_size
    }

    #[inline]
    fn get_entries_start(&mut self) -> *mut ACPISDTHeader {
        unsafe { (self as *mut Self).offset(1) }
    }

    #[inline]
    fn get_signature(&self) -> &str {
        unsafe { from_utf8_unchecked(&self.signature) }
    }

    fn compute_checksum(&self) -> u8 {
        let mut sum: u8 = 0;

        let mut start = self as *const Self as *const u8;

        unsafe {
            for _ in 0..self.length {
                sum = sum.wrapping_add(*start as _);
                start = start.offset(1);
            }
        }

        sum as _
    }
}

unsafe fn print_entries(rsdt_or_xsdt: *mut ACPISDTHeader, entry_ptr_size: usize) {
    info!(
        "xsdt or rsdt: signature {}, checksum : {}",
        (*rsdt_or_xsdt).get_signature(),
        (*rsdt_or_xsdt).compute_checksum()
    );

    let mut start = (*rsdt_or_xsdt).get_entries_start();
    let len = (*rsdt_or_xsdt).get_entries_len(entry_ptr_size);

    for i in 0..len {
        let a = (*start).signature;
        let length = (*start).length;

        info!("entry {i} : signature {a:?}, length {length}");
        start = start.byte_add(entry_ptr_size);
    }
}

#[entry]
fn main(_image_handle: Handle, mut system_table: SystemTable<Boot>) -> Status {
    uefi::helpers::init(&mut system_table).unwrap();

    for config_entry in system_table.config_table().iter() {
        match config_entry.guid {
            ACPI2_GUID => {
                let rsdp = config_entry.address as *mut RSDP;
                unsafe {
                    let xsdt = (*rsdp).xsdt_address as *mut ACPISDTHeader;
                    print_entries(xsdt, 8);

                    let rsdt = (*rsdp).rsdt_address as *mut ACPISDTHeader;
                    print_entries(rsdt, 4);
                }
            }
            _ => {}
        }
    }

    system_table.boot_services().stall(20_000_000);
    Status::SUCCESS
}
```
my results 

[image]

nicholasbishop, 2024-07-09:
I think you're missing a level of pointer indirection. `get_entries_start` should look something like this:
```rust
    #[inline]
    fn get_entries_start(&self) -> *const *const ACPISDTHeader {
        let start = unsafe { (self as *const Self).offset(1) };
        start.cast()
    }
```

Each entry is a pointer (either 32-bit or 64-bit depending on the table), so you need to deference a second time.

```rust
        let a = (*(*entry)).get_signature();
        let length = (*(*entry)).length;
```

With that change in place I can see what looks like valid XSDT data:
```
[ INFO]: uefi-test-runner/src/main.rs@088: xsdt or rsdt: signature XSDT, checksum : 0
[ INFO]: uefi-test-runner/src/main.rs@108: entry 0 : signature "FACP", length 244
[ INFO]: uefi-test-runner/src/main.rs@108: entry 1 : signature "APIC", length 144
[ INFO]: uefi-test-runner/src/main.rs@108: entry 2 : signature "HPET", length 56
[ INFO]: uefi-test-runner/src/main.rs@108: entry 3 : signature "MCFG", length 60
[ INFO]: uefi-test-runner/src/main.rs@108: entry 4 : signature "WAET", length 40
[ INFO]: uefi-test-runner/src/main.rs@108: entry 5 : signature "BGRT", length 56
```

You'll need to adjust the code a little for the 32-bit pointers.

Zekirax, 2024-07-11:
Thanks a lot. But for me "addresses that point to other DESCRIPTION_HEADERs" means that the pointers point to the a DESCRIPTION_HEADER and not a pointer that point to a DESCRIPTION_HEADER.

nicholasbishop, 2024-07-11:
You're correct, but the reason there's a double pointer involved is because the "outer" pointer is pointing to the list of pointers. So each inner pointer (the entries in the list) is indeed pointing to a `DESCRIPTION_HEADER`, and the outer pointer (the double pointer) only exists within the code, not part of the data structure as such.

```
TOP-LEVEL HEADER
entry_pointer -> HEADER  <-- get_entries_start() returns a pointer to this pointer 
entry_pointer -> HEADER
entry_pointer -> HEADER
```

closed 2024-07-11 by nicholasbishop