rust-osdev/uefi-rs #1241: Redesign Allocator API

radioactiveray, 2024-07-14:
Suggested Changes
-
- Add three different allocators: `LoaderAllocator`, `BootServicesAllocator` and `RuntimeServicesAlloctor`
- Add three different flags: `alloc_loader`, `alloc_boot_driver` and `alloc_runtime_driver`
- Set the global allocator to one of the three allocators depending on the flag whereas `loader` would be the default one
- Change the `uefi::helpers::init` to only initialize the logger if the feature is enabled
- Remove the Allocator initialization and deinitialization functions
- Note: I have only used an example naming convention

Reason
-
- Since `uefi-rs` 0.29.0 the `#[entry]` macro automatically sets the global `SystemTable`
- The whole allocation function includes an additional load of a static variable which introduces extra overhead
- The `MEMORY_TYPE` is defined by the Subsystem of the UEFI Image which is defined by the link arguments
- The link arguments need to be manually changed to get a boot or runtime driver so could it also be done manually with specifying the relevant flags

Idea
-
Changing this would remove some extra overhead to the initialization and would remove the need for any extra visible code extra than the `#[entry]` macro. In addition to this the allocation code would need minimal changes and a static variable could be removed.

Future
-
Changing this behavior would give the possibility to create one 'alloc' like function that would also take as extra argument the EFI_MEMORY_TYPE which would give the possibility to also add some sort of a custom allocator which could the user use the allocate data with special markings in the memory map after calling `exit_boot_services`.

Some Example Code
-
```

// Dummy implementation

#[derive(Debug)]
pub struct LoaderAllocator;

#[derive(Debug)]
pub struct BootServicesAllocator;

#[derive(Debug)]
pub struct RuntimeServicesAllocator;

unsafe fn alloc_impl(layout: Layout, memory_type: MemoryType) -> *mut u8 { /* snip */ } // Just don't retrieve the memory type and use the one provided
unsafe fn dealloc_impl(mut ptr: *mut u8, layout: Layout) -> *mut u8 { /* snip */ }

impl GlobalAlloc for LoaderAllocator {
    unsafe fn alloc(&self, layout: Layout) -> *mut u8 { alloc_impl(layout, MemoryType::LOADER_DATA) }

    unsafe fn dealloc(&self, ptr: *mut u8, layout: Layout) { dealloc_impl(ptr, layout) }
}

impl GlobalAlloc for BootServicesAllocator {
    unsafe fn alloc(&self, layout: Layout) -> *mut u8 { alloc_impl(layout, MemoryType::BOOT_SERVICES_DATA) }

    unsafe fn dealloc(&self, ptr: *mut u8, layout: Layout) { dealloc_impl(ptr, layout) }
}

impl GlobalAlloc for RuntimeServicesAllocator {
    unsafe fn alloc(&self, layout: Layout) -> *mut u8 { alloc_impl(layout, MemoryType::RUNTIME_SERVICES_DATA) }

    unsafe fn dealloc(&self, ptr: *mut u8, layout: Layout) { dealloc_impl(ptr, layout) }
}

// Depending on the flag

#[cfg(all(feature = "alloc_loader", any(feature = "alloc_boot_driver", feature = "alloc_runtime_driver")))]
#[global_allocator]
static ALLOCATOR: Allocator = LoaderAllocator;

// or

#[cfg(all(feature = "alloc_boot_driver", any(feature = "alloc_loader", feature = "alloc_runtime_driver")))]
#[global_allocator]
static ALLOCATOR: Allocator = BootServicesAllocator;

// or

#[cfg(all(feature = "alloc_runtime_driver", any(feature = "alloc_loader", feature = "alloc_boot_driver")))]
#[global_allocator]
static ALLOCATOR: Allocator = RuntimeServicesAllocator;

#[cfg(any(
    all(feature = "alloc_loader", feature = "alloc_boot_driver"),
    all(feature = "alloc_loader", feature = "alloc_runtime_driver"),
    all(feature = "alloc_boot_driver", feature = "alloc_runtime_driver"),
    all(
        feature = "alloc_loader",
        feature = "alloc_boot_driver",
        feature = "alloc_runtime_driver"
    )
))]
compile_error!(
    "only one of \"alloc_loader\", \"alloc_boot_driver\" or \"alloc_runtime_driver\" can be activated at the same time"
);

// This may be done via mutual exclusive features which normally should be avoided or some cfg_if code 

// A custom allocator

#[derive(Debug)]
pub struct CustomAllocator {
    memory_type: MemoryType,
}

impl CustomAllocator {
    pub const fn new(memory_type: MemoryType) -> Self {
        Self { memory_type }
    }
}

impl GlobalAlloc for RuntimeServicesAllocator {
    unsafe fn alloc(&self, layout: Layout) -> *mut u8 { alloc_impl(layout, self.memory_type) }

    unsafe fn dealloc(&self, ptr: *mut u8, layout: Layout) { dealloc_impl(ptr, layout) }
}
```

I know that mutual exclusive feature flags are not preferred but in this special case the programmer needs to know how everything works and would then know to only activate one feature flag

nicholasbishop, 2024-07-16:
Hi, thanks for writing this up. Could you summarize what problem or problems you want to solve? There's a lot going on here so it's a little hard for me to pick out what the main idea is :)

> The whole allocation function includes an additional load of a static variable which introduces extra overhead

Is this a problem in practice? Having a specific motivating example would be helpful, i.e. a project that uses uefi-rs and is having performance problems due to this static var.

radioactiveray, 2024-07-16:
TL;DR
-
- Change the situation that the `uefi::helper::init` function is necessary (maybe also set with `#[entry]`)
- Add a new alloc function (alloc_impl in my example code)
- Give the user the possibility to easily create their own allocator for a specific `MEMORY_TYPE`
- Add allocators for the known `EFI_MEMORY_TYPE` (not necessarily all)

Long answer
-
My general problem with the API is that the current situation requires everyone who wants to write a default UEFI Application or boot loader and not some driver needs to add this extra line of code, which after the release of 0.29.0 does only set the MEMORY_TYPE for the allocation as the global system table pointer is automatically set with the `#[entry]` macro.

I have no real world use case that is really limited by this behavior, despite some sort of micro optimizations that are potentially made as the static access may involve a cache fetch to the memory at which the static is stored but I may be wrong with this one.

The extra overhead I mentioned is that a static will involve some user code interaction as the `[no_std]` UEFI does not feature some sort of `LazyCell` which would automatically initialize at first access. This is no performance overhead but a simple code overhead not directly needed.

In addition to this would the usage of a more general `alloc` with a memory type as a parameter function provide an easier possibility in implementing custom allocator that will use different a different MEMORY_TYPE which is useful for identifying data in the `MemoryMap` after calling `exit_boot_services`.
With this I assume that the Firmware will automatically set the underlying page tables of the pool allocation to the corresponding `EFI_MEMORY_TYPE` in the `MemoryMap` .

Overall a change would give the possibility to remove the `uefi::helpers::init` function which would give the user a more clear structure when building a simple application. So I did not meant some real world performance problem but rather a general problem when usage of a static that needs to be initialized by the user. That means a slight change in the API with an `unsafe fn alloc_impl(layout: Layout, memory_type: MemoryType) -> *mut u8` with a public interface or some other Allocator may could help other projects to easier allocate some special resources depending on their needs, as for example when a boot loader needs to allocate memory for the os that uses a MEMORY_TYPE in the OS Range.

Alternatives
-
For an easier usage it would also be possible to set the `MEMORY_TYPE` to `MemoryType::LOADER_DATA` and then add functions to change this. But these function would be extremely unsafe as they could break everything when called after an allocation.

radioactiveray, 2024-07-16:
Example Usage
-
An Allocator with a custom `EFI_MEMORY_TYPE` could be used by a boot loader to generate a device-structure-graph-like in memory structure, which would be hard to create when only using page allocation, and then would be some sort of marked in the `MemoryMap` and could be identified by the kernel or some Page Frame Allocator after `exit_boot_services`. After this a new kernel allocator could be used to copy the structure to some new location with a new kernel allocator, what would make it possible to reuse the memory with the custom `EFI_MEMORY_TYPE`.
Without a custom allocator this would be significantly harder to program and maintain when a device-graph would take up multiple pages of memory allocated with pool memory.

phip1611, 2024-07-21:
I personally see the proposed changes with various possible allocators extended as Cargo features as complication and unnecessary sophistication. I'd much rather extend the `uefi::helpers::init` function to be more flexible, such as configuring a default allocation memory type there.

radioactiveray, 2024-07-21:
I understand your points and I also personally rethinked about the effect of a change in the last days and came to the point it is no real or effective benefit to the system, therefore I will close this issue.

closed 2024-07-21 by radioactiveray