rust-osdev/uefi-rs #1303: Evaluate, Discuss + Document role of "alloc" feature in Crate

phip1611, 2024-08-09:
As pointed out in https://github.com/rust-osdev/uefi-rs/pull/1297#issuecomment-2278207759, we should discuss our strategy regarding the `alloc` feature and its priority when creating user API. This should also be documented, once we discussed and agreed on a strategy.

phip1611, 2024-08-09:
So far, the used approach is unspoken and was never really discussed. From my point of view, it was a 66.6% prefer non-alloc and a 33.3% do as you like.

nicholasbishop, 2025-01-26:
Making a note from https://github.com/rust-osdev/uefi-rs/pull/1528: we also have several types for data allocated internally with UEFI pool memory: `PoolString`, `PoolDevicePath`, PoolDevicePathNode`. That avoids extra memory allocations for some code, but unclear that the extra API surface is worth that optimization.