rust-osdev/uefi-rs #1320: Use with nightly uefi-std

Aalivexy, 2024-08-12:
Can this crate be used with nightly uefi-std? Is there an example?

phip1611, 2024-08-12:
Excellent question. Actually, this is something that also came to my mind previously. We want to write something about this in https://github.com/rust-osdev/uefi-rs/pull/1292/commits/979c1311a99a7dfba5018ca3fc523a8c1edbcf7e - but we need to investigate it.

phip1611, 2024-08-14:
Not sure if it will be merged in that form, but this is a minimal example @Aalivexy  https://github.com/rust-osdev/uefi-rs/pull/1292/commits/5dc4386f9a0ccdc4be8bddf9984dfda0ec78a518

Aalivexy, 2024-08-14:
> Not sure if it will be merged in that form, but this is a minimal example @Aalivexy  https://github.com/rust-osdev/uefi-rs/pull/1292/commits/5dc4386f9a0ccdc4be8bddf9984dfda0ec78a518

Thank you very much for your help! However, since it is still a nightly feature (in fact, only getting uefi tables is a nightly feature), it may be better to wait for it to stabilize.

closed 2024-08-15 by nicholasbishop