rust-osdev/uefi-rs #1438: The input protocol dosen`t work on real keyboards.

Motanescu1357, 2024-10-21:


phip1611, 2024-10-21:
Hi! Please provide more information about the code you ran, the environment (software versions etc.) you used, and so on. Without context, we can't do much about it.

Motanescu1357, 2024-10-21:
The Input protocol does not get user input on a real computer.

phip1611, 2024-10-21:
FYI: I slightly edited your reply to use proper [Markdown](https://docs.github.com/en/get-started/writing-on-github/getting-started-with-writing-and-formatting-on-github/basic-writing-and-formatting-syntax).

Motanescu1357, 2024-10-21:
Maybe the problem is in the Input protocol.
When I compile the project and test the .efi  file with a vm it works.

nicholasbishop, 2024-10-21:
This line doesn't look right to me:
```rust
input_protocol.wait_for_key_event().unwrap();
```

This is just getting an `Event`, it's not actually waiting for a key to be pressed (see https://docs.rs/uefi/latest/uefi/proto/console/text/struct.Input.html#method.wait_for_key_event). You'll need to use [`boot::wait_for_event`](https://docs.rs/uefi/latest/uefi/boot/fn.wait_for_event.html) to actually wait for a key press.

Motanescu1357, 2024-10-21:
It sill does not work.

nicholasbishop, 2024-10-21:
Can you describe in what way? Is it panicking somewhere? Does the `boot::wait_for_event` call ever return? Is `key` ever `Some`?

Motanescu1357, 2024-10-21:
From `boot::wait_for_event` I get 0 and `key` is not `Some`.

Motanescu1357, 2024-10-21:
The `Input` protocol does not work in a real pc.

Motanescu1357, 2024-10-22:
The problem probably is the `Input` protocol not getting the pressed key.

Motanescu1357, 2024-10-24:
The issue is still active.

phip1611, 2024-10-24:
> The issue is still active.

We are all maintaining this in our free time 😉 You can be sure that @nicholasbishop and I have seen this issue. But at this time, none of us had the time to investigate this further, unfortunately.

nicholasbishop, 2024-10-26:
I think I've identified the issue. I trimmed your example code down and verified what you saw: key presses worked in a VM, but not on real hardware. The device I tested on is a Lenovo X1C gen 9. The issue is this line: `system::with_stdin(|stdin| stdin.reset(true).expect("Failed to reset stdin"));`. Once I removed that, it started working.

Here's the complete code I tested:

```rust
// In cargo.toml:
// uefi = { version = "0.33.0", features = ["panic_handler"] }

#![no_main]
#![no_std]

use uefi::prelude::*;
use uefi::println;
use uefi::proto::console::text::Input;

#[entry]
fn main() -> Status {
    uefi::helpers::init().unwrap();

    let handle = boot::get_handle_for_protocol::<Input>().unwrap();
    let mut input_protocol = boot::open_protocol_exclusive::<Input>(handle).unwrap();

    loop {
        println!("before wait");
        boot::wait_for_event(&mut [input_protocol.wait_for_key_event().unwrap()]).unwrap();
        println!("after wait");
        let key = input_protocol.read_key().unwrap();
        println!("key: {key:?}");
    }
}
```

Motanescu1357, 2024-10-28:
Thanks, I turned your code into the input processing code and pasted my own code for handling  commands in.

closed 2024-10-28 by Motanescu1357