rust-osdev/uefi-rs #1453: println macro dosen`t work.

Motanescu1357, 2024-10-28:


phip1611, 2024-10-28:
Can you please be more specific about the environment, your setup, your code?

Have you read the [documentation](https://docs.rs/uefi/latest/uefi/macro.print.html#usage)?

> Use this similar to print! from the Rust standard library, **but only as long as boot services have not been exited.**

Does that help?

Motanescu1357, 2024-10-28:
I did not run my code in a vm.
I atempted to print a `String` to the screen.
The print macro does not print to the screen.
I did not exit boot services.
Here is my code:
rust```
#![no_main]
#![no_std]
extern crate alloc;
use uefi::prelude::*;
use uefi::{print, println, CStr16};
use uefi::proto::console::text::{Input, ScanCode};
use uefi::proto::console::text::Key;
use alloc::string::{String, ToString};
use uefi::proto::media::file::{File, FileAttribute, FileMode};
use uefi::runtime::ResetType;

#[entry]
fn main() -> Status {
    uefi::helpers::init().unwrap();
    println!("This is a small project that I made.");
    println!("Use The escape key or the exit command to shutdown the system.");
    print!(">");
    let handle = boot::get_handle_for_protocol::<Input>().unwrap();
    let mut input_protocol = boot::open_protocol_exclusive::<Input>(handle).unwrap();
    let mut input:String = String::new();
    loop {
        boot::wait_for_event(&mut [input_protocol.wait_for_key_event().unwrap()]).unwrap();
        if input == "exit" {
            runtime::reset(ResetType::SHUTDOWN, Status::SUCCESS, None);
        }
        let key = input_protocol.read_key().unwrap();
        let mut char:String;
        if let Some(Key::Printable(char2)) = key {
            char = char2.to_string();
            if char != "\r" {
                println!("{}", char);
                input.push_str(&char.as_str());
            }else{
                char = char2.to_string();
                os(char)
            }
        }
        if let Some(Key::Special(scancode)) = key {
            if scancode == ScanCode::ESCAPE {
                runtime::reset(ResetType::SHUTDOWN, Status::SUCCESS, None);
            }
            if scancode == ScanCode::DELETE {
                input.pop();
            }
        };
    }
}
fn os(input:String) {
    if input.contains("help") {
        println!();
        println!("Commands:");
        println!("    exit: Exits the system");
        println!("    remove [<target_path>]: The command removes the target.");
        println!("    new-file [<target_path>]: A new file at the target directory.");
        println!("    new-dir [<target_path>]: A new directory at the target directory.");
        println!("    edit-file <file_contents> [<target_path>]: Edits a file and ads the file_contents to the file at the target directory.");
        println!("    copy [<file1_path>] | [<file2_path>]:Creates a copy of the file.");
        println!("    cat [<target_path>]:Prints the text in a file");

    }
    if input.contains("remove") {
        let mut filesystem = boot::get_image_file_system(boot::image_handle()).unwrap();
        let buf= &mut [0; 99999];
        let binding = input.replace("remove", "").replace(" ", "").replace("/", "\\");
[... 77 lines cut ...]
                let mut file1_contents:&mut [u8] = &mut [];
                let _ = file1.read(&mut file1_contents).unwrap();
                let _ = file2.write(&file1_contents);
                file1.delete().unwrap();
                file2.close();
            }
        }
    }
    if input.contains("cat") {
        let mut filesystem = boot::get_image_file_system(boot::image_handle()).unwrap();
        let buf = &mut [0; 999];
        let binding = input.replace("cat", "").replace(" ", "").replace("/", "\\");
        let str = binding.as_str();
        let file_path = CStr16::from_str_with_buf(str, buf).unwrap();
        let file = filesystem.open_volume().unwrap().open(file_path, FileMode::Read, FileAttribute::default()).unwrap();
        if file.is_regular_file().unwrap() == true {
            let file_contents:&mut [u8] = &mut [];
            file.into_regular_file().unwrap().read(file_contents).unwrap();
            println!("File contents: {:?}", String::from_utf8(file_contents.to_vec()).unwrap());

        }
    }
}
```
I removed a `prtintln` macro because it contained my name.

Motanescu1357, 2024-11-03:
No, `println!` does print a line to the screen.

Motanescu1357, 2024-11-03:
I made the input string actually work and now `print!` actually works but I have problems with implementing the enter and backspace keys.

closed 2024-11-03 by Motanescu1357